travisgoodspeed/md380tools #1: Firmware Updates

travisgoodspeed, 2015-09-26:
Firmware updates come in an encrypted form, but they are decrypted by the radio during the loading process.  `md380-dfu` should be able to send the encrypted update, letting the radio take care of the decryption.

Close this issue when the `md380dfu` can flash firmware

mach327, 2016-01-31:
I may have something for this, but I learned everything I know about USB, DFU, and a fair amount of python in getting it to work, so I have little confidence my code is worth opening a pull request for. I also made lots of changes in working to understand everything.

But, it does work! (on my machine and radio).
I also have PCAPs of the original upgrade.exe pushing new firmware to the radio I can share on request.

download_firmware(): https://github.com/mach327/md380tools/blob/master/md380-dfu#L365

travisgoodspeed, 2016-01-31:
Thanks @mach327, I confirm that your fork works to reflash the radio.  I've been using the DFU bootloader for development, but this will conveniently remove that as a requirement.

Re-assigning this issue to myself for merging, which should be finished tonight or tomorrow.

73 from Knoxville,
--Travis

commit a6decdb4 (2016-01-31): Manual merge of a patch from @mach327 for flashing firmware from md3800-dfu.  Close #1.
closed 2016-01-31 by travisgoodspeed