travisgoodspeed/md380tools #4: Chirp Timeslot

travisgoodspeed, 2015-10-01:
Chirp is presently ignoring the timeslot.  Close this issue when it can be configured.

mach327, 2016-05-02:
May have something for this soon-ish.

I can successfully read and write the timeslot and color code from within Chirp with this commit https://github.com/mach327/chirp_fork/commit/e61c992acc6ee3886f906a348de7a5677472ea83
(I tried to emulate the current DStar support in Chirp to get DMR support.)

and the [current MD380 driver in my md380tools repo ](https://github.com/mach327/md380tools/blob/master/chirp/md380.py)

[image]

travisgoodspeed, 2016-05-02:
Nifty!  How much trouble do your think it will be to add Contacts and Listen Groups to CHIRP?

mach327, 2016-05-02:
It seems like it will be more complex than the current D-Star support, which is around 200 lines of code larger than the current DMR support. I haven't closely read and understood all the D-Star code, but that's my next step.

rct, 2016-05-03:
@mach327 - It might be worth mentioning your work on chirp_devel@intrepid.danplanet.com to see if you can get any help/insights from any of the regular chirp developers that might have some interest in DMR.

mach327, 2016-05-04:
Good idea, I'll do that.

travisgoodspeed, 2016-07-08:
Closing stale CHIRP issues in favor of #41.

closed 2016-07-08 by travisgoodspeed