travisgoodspeed/md380tools #8: DMR-MARC Bulk Import

travisgoodspeed, 2015-10-02:
Chirp data sources for DMR-MARC's data dump would allow repeaters to be automatically loaded.
http://www.dmr-marc.net/cgi-bin/trbo-database/datadump.cgi

thorsen999, 2016-02-01:
There is a software that can be used in parallel with the Tytera CPS to bulk programm the md380 here : 
http://dl5mcc.de/cpsprogrammer/index.php
It uses a couple of links to generate csv files with the channels, zones and scanlists and manipulate tytera software to do the bulk programming.
Hope it helps ;-) 

Tom F8COD

mach327, 2016-02-05:
I've been poking at adding DMR-MARC database fetching and filtering to Chirp, but not functional just yet. Have it filtering by state, want to have it filter by country and band also. Haven't convinced Chirp to parse the CSV columns properly yet either, I need to read a lot more code.
I may have more time to prod it into submission this weekend, but we'll see.

travisgoodspeed, 2016-02-24:
This is now done.  You can try it out by installing the hooked firmware (`make clean flash` when PTT and the button above it are held) and then running `md380-tool callog`.

```
tmg% md380-tool

Usage: md380-tool <command> <arguments>

Print a log of incoming DMR calls to stdout.
    md380-dfu calllog

Looks up the name by an ID number.
    md380-dfu lookup 12345

tmg% md380-tool calllog
DMR call from KK4VCZ Travis Goodspeed (3147092) to 99.

```

closed 2016-02-24 by travisgoodspeed

ajorg, 2016-02-24:
Neat!