travisgoodspeed/md380tools #10: block_size > 4096

iddq, 2015-11-17:
what is the usb backend that you use for pyUSB that can handle larger usb transfer size than 4096 bytes?

travisgoodspeed, 2015-11-17:
libusb-1.0

closed 2015-11-17 by travisgoodspeed

iddq, 2015-11-22:
have you modified the libusb? MAX_CTRL_BUFFER_LENGTH is 4096

libusb/os/linux_usbfs.h:#define MAX_CTRL_BUFFER_LENGTH          4096

./libusb/os/linux_usbfs.c:

        if (transfer->length - LIBUSB_CONTROL_SETUP_SIZE > MAX_CTRL_BUFFER_LENGTH)
                return LIBUSB_ERROR_INVALID_PARAM;

travisgoodspeed, 2015-11-23:
I used libusb-1.0 on a Mac, but I didn't encounter this issue.  Please get in touch with me by email, travis at radiantmachines, and I can send you longer form notes without cluttering github issues.