travisgoodspeed/md380tools #17: Forward port public/private patches to firmware 2.034

rct, 2016-02-04:
Forward port the "scanning" patches, listen to all talkgroups and all private calls, to the current firmware version D2.034.

The current patches are against firmware version 2.032.

Issue dded for tracking/documentation purposes.

travisgoodspeed, 2016-02-04:
Is a change log available from Tytera that describes what's new in 2.034?

selabnayr, 2016-02-04:
http://www.va3xpr.net/programming-software-firmware/ has a good set of firmwares, and it looks like the Operation.doc.pdf file in the archives has a changelog.

selabnayr, 2016-02-04:
Ah, changelog format is a bit weird. The bolded section is what's new since the previous release.

travisgoodspeed, 2016-02-04:
So the new features in bold are as follows:
```
13. Correct spelling of repeatar in Utilities-program radio-repeatar to repeater
14. Solve the wrong display when deleting entry from Missed List. When the
last entry is deleted the screen displays:“??? ??? ?”
15. Solve the problem when we program the radio, there will be some data missing.
```

I'm pushing a patch for 13 in a minute.  Are 14 or 15 worth the trouble of porting the scripts and abandoning the current symbols, or should we wait another version?

rct, 2016-02-04:
@selabnayr thanks for the clarification.
@travisgoodspeed - maybe not, though would like to find out what _"15. Solve the problem when we program the radio, there will be some data missing."_ really means.

travisgoodspeed, 2016-02-04:
2.032 has some really janky bit-bang code to read and write the SPI Flash.   I'm guessing that it gave them trouble and they made a minor change.

I'm leaving this issue open for now, because we will need to be version-agnostic in the future, but for now I'll continue to use absolute addresses.  I'll go back to clean things up after the LCD display is better understood.

travisgoodspeed, 2016-03-08:
So I really, truly wanted to label this as WONTFIX and call it a day, but it looks like we need 2.034 to support the MD446.  Version 2.032 will flash to the MD446, but the keypad is incompatible and the menu items are unreachable.

I'll buy a six pack of good beer for whomever can automate the symbol transition from 2.032 to 2.034, even if a few symbols have to manually patched up afterward.

travisgoodspeed, 2016-03-26:
Looks like 2.034 is built separately for the MD446 and MD380, so there won't be one version that's compatible with both.  Moving this back to wontfix.

closed 2016-03-26 by travisgoodspeed