travisgoodspeed/md380tools #18: Chirp driver upload and download

mach327, 2016-02-05:
I filled out some of the Chirp driver to import md380-dfu and run upload_codeplug and download_codeplug. I had to patch Chirp to ignore some serial port stuff, and added a NO_SERIAL "flag" to the Radio driver. I don't know if it's the cleanest or sanest way to make Chirp understand, but it'll do for testing for now at least.

Patch for Chirp (`hg diff` output, anyway):
https://github.com/mach327/md380tools/blob/master/chirp_md380_NO_SERIAL.patch

Current md380 driver for Chirp:
https://github.com/mach327/md380tools/blob/master/chirp/md380.py

None of this is ready to be merged or used for anything but development - but it does successfully upload and download codeplugs to/from the radio. Someone will have to decide how Chirp will include the MD380 stuff eventually, if the driver will be distributed with Chirp. 

In the meantime, this can be tested and worked on by symlinking the chirp driver `chirp/md380.py` into Chirps `drivers/` directory, and symlinking the md380tools repository into {whichever python your using}'s site-packages directory (or wherever else you like in your python path).
For future hg noobs like me, looks like you can apply the chirp patch on your cloned Chirp repo to play with asking like so:
`hg patch ../path/to/chirp_md380_NO_SERIAL.patch --no-commit`.

rct, 2016-02-07:
FYI, the DCS (digital coded squelch) case on analog channels isn't being handled properly.  Chirp throws an error.  (DCS 503 N) is getting tranlated into a CTCSS value of 850.3.   I probably should open a separate issue.   I didn't check @mach327's repo yet.

rct, 2016-02-07:
Also, another Chirp patch to the main code that will be needed is to add .rdt to the file type list so you can open .rdt wrapped code plugs.  The code already detects the difference between .img and .rdt based upon lengths to get the right offset into the memory map.

```
diff -r 93e725a0c16e chirp/ui/mainapp.py
--- a/chirp/ui/mainapp.py       Sun Jan 31 10:08:41 2016 -0800
+++ b/chirp/ui/mainapp.py       Sun Feb 07 12:02:17 2016 -0500
@@ -323,6 +323,7 @@
                      (_("VX5 Commander Files") + " (*.vx5)", "*.vx5"),
                      (_("VX6 Commander Files") + " (*.vx6)", "*.vx6"),
                      (_("VX7 Commander Files") + " (*.vx7)", "*.vx7"),
+                     (_("CS700/MD380 Codeplug Files") + " (*.rdt)", "*.rdt"),
                      ]
             fname = platform.get_platform().gui_open_file(types=types)
             if not fname:
@@ -785,7 +786,9 @@
                  (_("Travel Plus Files") + " (*.tpe)", "*.tpe"),
                  (_("VX5 Commander Files") + " (*.vx5)", "*.vx5"),
                  (_("VX6 Commander Files") + " (*.vx6)", "*.vx6"),
-                 (_("VX7 Commander Files") + " (*.vx7)", "*.vx7")]
+                 (_("VX7 Commander Files") + " (*.vx7)", "*.vx7"),
+                 (_("CS700/MD380 Codeplug Files") + " (*.rdt)", "*.rdt"),
+        ]
         filen = platform.get_platform().gui_open_file(types=types)
         if not filen:
             return
```

travisgoodspeed, 2016-02-07:
Yup.  We'll also need support during export, adding DFU wrapper or not as appropriate.

The use of raw (headerless) images from the SPI Flash was handy during reverse engineering of the format, but it's probably best to stick entirely to the RDT format when we merge support into Chirp.

mach327, 2016-09-24:
For RDT support, #241 
Otherwise closing in favor of the more active, more general #41

closed 2016-09-24 by mach327