travisgoodspeed/md380tools #21: Read/Write radio's time and date

rct, 2016-02-13:
The vendor software, MD_380, has operations to read or write the radio's date and time setting.

md380tools should have an option to set the radio's time and date using the host's current time.

I'm putting this in as a place holder for now.  Currently, I don't have a windows VM built yet that would let me trace the USB operations with VMware.  (Need to find out if USB tracing is also available from VirtualBox.)

Notes:
* Using MD_380 window software, connect the radio, select Edit -> Time Setting.  This brings up a dialog box with buttons to read or write the radio's time and date.
* The radio briefly goes into "PC Progam USB mode", and then returns to normal operation.

travisgoodspeed, 2016-02-14:
Yup, this should be added.  Luckily, it's a nice short transaction that'll stand out in a pcap.

aeickho, 2016-02-20:
pcap file of 
write time: https://flauschfunk.de/write_date_time.pcapng
read time: https://flauschfunk.de/read_date_time.pcapng

aeickho, 2016-02-21:
on read time: https://flauschfunk.de/read_date_time.pcapng
the  date/time is in frame 245

travisgoodspeed, 2016-02-21:
Thanks!  Sticking to `read_date_time.pcapng` for this comment.

Yup, that's the date in 7-bytes of BCD returned in packet 245.
[image]

Packet 244 is a normal UPLOAD request, for 7 bytes from block 2.  (DFU reads and writes index from 2 within a memory.  0 is used for control and 1 for our non-standard patches.)

Working backward, we ignore 241/242 as a STATUS pair.

228 and 229, however, are more promising.  They are a DFU secret command pair of 0xA2 0x08, and I'm pretty sure from reading the firmware that the A2 commands all select a memory target.

Sure enough, this little example will dump the current time from the device.
```
                import usb.core
                dfu = init_dfu()
                dfu.md380_custom(0x91,0x01); #Programming Mode                                                                                                                                                             
                dfu.md380_custom(0xA2,0x08); #Access the clock memory.                                                                                                                                                     
                time=dfu.upload(0,7);  #Read the time bytes as BCD.                                                                                                                                                        
                hexdump(time);
```

```
iMac% md380-dfu time
20160220 194440
iMac% 
```

Writing in this same way is probably enough to set the clock.  I'll toss together a test now and merge it if it works.

travisgoodspeed, 2016-02-21:
In the write PCAP, working backward from the end,

Packet 248 performs the reboot by calling 0x91,0x05 as a secret command.

Packet 231 performs a write of eight bytes beginning with B5 and followed by the 7-byte BCD date.  Trying that.

Packet 213 send 0x91,0x02.

Packet 179 sends the 0xA2, 0x07 sequence.

Packet 145 sends 0xA2, 0x04.

Packet 111 sends 0xA2, 0x03.

Packet 77 sends 0xA2, 0x02.

Packet 43 sends 0xA2, 0x01.

Packet 27 sends 0x91, 0x01, entering programming mode.

From the firmware, I recall that a lot of the A2 modes have to go in order.  I'll try rigging up a client to do them in sequence early next week.

travisgoodspeed, 2016-02-21:
I merged your pull request for the raw read of the time in md380-tool.  What's the procedure for a raw write?

aeickho, 2016-02-22:
i work on it

aeickho, 2016-03-17:
usb_upld_handle() @ 0x0808d448 (2.032) there is the command "8" parser  :)

ea, 2016-11-29:
Should this be closed or is there anything else that needs to be done ?

travisgoodspeed, 2016-11-29:
@ea I'll try it out now and close the issue if there's no trouble.

travisgoodspeed, 2016-11-29:
The pull request broke `md380-dfu time`, but with a4a57d1, the regression is fixed.

Happily closing one of our oldest issues.

closed 2016-11-29 by travisgoodspeed