travisgoodspeed/md380tools #22: Access to radio's calibration data.

rct, 2016-02-13:
Interesting find that I just saw reported on the MD_380 Yahoo group.  The MD_380 windows application has the ability to read and write the radio's calibration data,   The software will save the settings in an 8 KB binary blob with a `.test` extension.

To access this, you need to change the line `testmode=0` to 1 in `setting.ini`.

The path on a default Windows 7 install is `C:\Program Files (x86)\MD_380\MD_380\SoftWare_EN\setting.ini`

The `testmode=1` line is in the `[setup]` group of the `.ini` file.

The process is:

1. Change the settings.ini file as described above.
2. connect the radio
3. Start MD-380.exe
4. Hit Control-T which will read the data off the radio.
5. Click "Save Test Data" which will save a .test file with an 8 KB binary of the calibration data. 

It might be worth adding a note to the README to backup the radio's calibration data before doing too much experimentation.

I can post a screenshot if anyone can't run the Windows software.

rct, 2016-02-13:
I figured out the organization of the MD-380 .test file.  I should be able to through together a chirp bitwise structure for it when I get a chance.   While the .test file is 8 KB, only the first 512 bytes has data in it. the rest is 0xFF padding.  If they follow the same pattern that is probably the way the bits are organized on the flash.

Conveniently the data seems to be organized in 16 byte rows even those most of the calibration rows are only using 9 values.

The 9 columns seem to correspond to a set of 9 frequencies for calibration 401,411,421,...,479.99 mhz 

In the meantime, I wrote the details of the .test file down here while I was figuring it out http://rctnotes.blogspot.com/2016/02/structure-of-tytera-md-380-test-radio.html

travisgoodspeed, 2016-02-14:
Thanks!  That should be sufficient to read out the calibration data. 

In the firmware, I see a lot of strange DFU commands interacting with the OTP Flash.  Perhaps that's where it's stored?  In any case, packet sniffing will tell.

rct, 2016-02-14:
The window software lets you write the changed "test" settings back to the radio, so I'm guessing not OTP...

[screenshot](http://2.bp.blogspot.com/-MB4j7z3XX7A/Vr-XH4rBSAI/AAAAAAAAJYc/7yVcDLyxMCY/s1600/29gqvm3r.bmp)

aeickho, 2016-02-20:
pcap file from read calib. data:

https://flauschfunk.de/read_calib.pcapng

aeickho, 2016-03-27:
working hypothesis: radio's calibration data are stored in 3X256-Bytes Security Registers on spi flash

travisgoodspeed, 2016-03-28:
I'll add support for reading that over USB with md380-tool.

travisgoodspeed, 2016-11-30:
Removing my assignment.  With a little luck, @ea will have a pull request for us soon.

aeickho, 2016-12-12:
added with: https://github.com/travisgoodspeed/md380tools/commit/b0c01666484e2054d30994590cc6eab876f20b4c

closed 2016-12-12 by aeickho

narspt, 2017-04-23:
Sorry for the off-topic but anyone knows what's the difference for CPS having testmode=0 or testmode=1?
I know that all pages over internet say to change it to access test mode but apparently I can access the test mode window (and it seems fully working) even with testmode=0, any one knows what that setting does exactly?

UB1AAM, 2017-04-25:
Ctrl+T

narspt, 2017-04-25:
Yes... but Ctrl+T works even with testmode=0... even on older 1.30 CPS... then I'm curious why anybody suggests changing it... what the change does exactly...

travisgoodspeed, 2017-04-25:
This is an issue tracker, not a forum.  Please move the discussion to IRC or the mailing list.