travisgoodspeed/md380tools #26: Radio unique ID and/or serial number.

rct, 2016-02-23:
The MD-380 CPS software shows a Unique Device ID and/or serial number that is stored in the code plug (.rdt) header but doesn't seem to be in the code plug image contents AFAICT.   Might be read from the radio as part of a different transaction?

Notes:

* serial number for my radio is blank, but unique device ID is filled out.
* some code plugs (.rdt) that I've seen have serial numbers, but may be from (older?) radios?
* The unique device ID is at bytes 0x16D - 0x178 of the .rdt file in two parts
  * The part that is displayed first as '257505Q' is stored at 0x172-0x178, as bytes in reverse order (Q505752).
  * The part after the letter seems to be stored as a 40 bit (!?) integer at 0x16D - 0x171.

rct, 2016-07-08:
So I haven't been paying close attention lately.  Is this data available from one of the md380 transactions?  (if you wanted to create an .rdt code plug that would match the vendor's software?)

travisgoodspeed, 2016-07-09:
Apologies.  I was trying to clear stale github issues and closed this one by mistake.

marrold, 2017-04-17:
Would this help with users flashing the wrong firmware on their radio? Hopefully the serial number or unique ID correlates in GPS or non GPS models.

travisgoodspeed, 2017-04-18:
A few notes from IRC with Phr3ak, who is also playing with the serial numbers.

The USB device serial number seems to be laser-etched in STM32's mask ROM starting at 0x1FFF7A10.  This is after the bootloader ROM.

Reference code for reading this serial number is available in `usb_desc.c` in many of the STM32Cube examples like the following:

```
/**                                                                                                                                                                            
  * @brief  Create the serial number string descriptor                                                                                                                         
  * @param  None                                                                                                                                                               
  * @retval None                                                                                                                                                               
  */
static void Get_SerialNum(void)
{
  uint32_t deviceserial0, deviceserial1, deviceserial2;

  deviceserial0 = *(uint32_t*)DEVICE_ID1;
  deviceserial1 = *(uint32_t*)DEVICE_ID2;
  deviceserial2 = *(uint32_t*)DEVICE_ID3;

  deviceserial0 += deviceserial2;

  if (deviceserial0 != 0)
  {
    IntToUnicode (deviceserial0, (uint8_t*)&USBD_StringSerial[2] ,8);
    IntToUnicode (deviceserial1, (uint8_t*)&USBD_StringSerial[18] ,4);
  }
}
```

I've not checked whether this is the same serial number that is used in the codeplug file.

marrold, 2017-04-20:
Thanks Travis