travisgoodspeed/md380tools #42: MotoTRBO RAS

travisgoodspeed, 2016-03-02:
MotoTRBO RAS (Restricted Access System) avoids out current sniffing.  It would be nice to understand the system and to sniff its audio.

Forked from a request by @BabaYog in Issue #19.

the-sPrinGfield, 2016-04-14:
Mototrbo Packers with RAS basically fail the CRC check as i understand it. Maybe it is possible to "not care CRC"

travisgoodspeed, 2016-10-17:
Closing this as no progress has been made since March.  Add it to #370 if there's interest.

closed 2016-10-17 by travisgoodspeed

s58sms, 2022-01-27:
Hello,

any options to open this issue/thread again?

iScottybotty, 2022-03-01:
> Mototrbo Packets with RAS basically fail the CRC check as i understand it. Maybe it is possible to "not care CRC"
> 
> Typical AMBE LC Block
> 
> Non RAS
> data,data,data,CRC
> 1234,1234,1234,FEDC
> 
> RAS
> 
> data,data,data,RASRASRAS,CRC
> 1234,1234,1234,999999999,FF11
> sent packet
> data,data,data,CRC
> 1234,1234,1234,FF11
> 
> The CRC no longer matches. since the CRC dosnt match -  the radio ignores the packet and there is no rx unmute.
> 

Is there any further work been made on RAS for the 380? Would the above work? Could that "ignore CRC" entry work if coded correctly? Remember, this is to RX only, not TX. How critical is the CRC check, and can it be bypassed?

onemikedelta, 2022-03-03:
If someone able to understand the whole system. It is definitely beneficial to implement for tx also. I have motorola radios (tier2), sdr and both mmdvm repeater and hotspot, but unfortunately not motorola repeater atm. I will gladly help to test this feature.

7AAZ, 2022-03-26:
Travis ! give a little help. with this option is really missing (PLEASE)

dkscan, 2022-03-27:
The low level DMR stuff are probably implemented in the DMR hardware and not part of the firmware and most likely cannot be changed.

iScottybotty, 2022-03-27:
The GD77 firmware hack completely ignores RAS, whilst I haven't looked at the chips of both radios, I don't expect it to be very difficult. 🤷‍♂️

s58sms, 2022-03-31:
This has nothing to do with hardware. The radio has to ignore crc header. OPENGD77 ignores ras decoder, but only older firmwares. New one are focusing on HAM.

iScottybotty, 2022-03-31:
Awesome. So you fork it, make the necessary change, (ignore the CRC Header) and then ping it to Travis to release. Happy days.  

Scott.

s58sms, 2022-09-14:
Hello, is there any news about RAS support on md380 toolz?

iScottybotty, 2022-09-14:
Need the developer of the GD77 to come forward to code the bypass of the CRC check, or anyone else that’s capable. Until then, I guess no. But that’s what’s needed. 👍

Scott.