travisgoodspeed/md380tools #44: which task does the I2C IC U307

aeickho, 2016-03-04:
2.032
 
40005400  is "I2C1 I2C_CR1"

```
          0x0804bf4c      5ff40011       movs.w r1, 0x200000                                        
          0x0804bf50      2948           ldr r0, [pc, 0xa4]          ; [0x804bff8:4]=0x40005400  ;  
          0x0804bf52      faf7dffc       bl 0x8046914                ;[2]     

/ (fcn) fcn.08046914 14                                                                               
|           0x08046914 b    dff83023       ldr.w r2, [pc, 0x330]       ; [0x8046c48:4]=0x200011ac     
|           0x08046918      1060           str r0, [r2]                                               
|           0x0804691a      dff83003       ldr.w r0, [pc, 0x330]       ; [0x8046c4c:4]=0x200011b0     
|           0x0804691e      0160           str r1, [r0]                                               
\           0x08046920      7047           bx lr                  

 804691a:       f8df 0330       ldr.w   r0, [pc, #816]  ; 0x8046c4c
 8046974:       48b5            ldr     r0, [pc, #724]  ; (0x8046c4c)
 8046c4c:       11b0            asrs    r0, r6, #6
$grep 8046c48 od
 8046914:       f8df 2330       ldr.w   r2, [pc, #816]  ; 0x8046c48
 8046a12:       488d            ldr     r0, [pc, #564]  ; (0x8046c48)
 8046a1c:       488a            ldr     r0, [pc, #552]  ; (0x8046c48)
 8046a36:       4884            ldr     r0, [pc, #528]  ; (0x8046c48)
 8046a44:       4880            ldr     r0, [pc, #512]  ; (0x8046c48)
 8046a62:       4879            ldr     r0, [pc, #484]  ; (0x8046c48)
 8046a70:       4875            ldr     r0, [pc, #468]  ; (0x8046c48)
 8046a8c:       486e            ldr     r0, [pc, #440]  ; (0x8046c48)
 8046a9a:       486b            ldr     r0, [pc, #428]  ; (0x8046c48)
 8046aa8:       4867            ldr     r0, [pc, #412]  ; (0x8046c48)
 8046ac2:       4861            ldr     r0, [pc, #388]  ; (0x8046c48)
 8046ad0:       485d            ldr     r0, [pc, #372]  ; (0x8046c48)
 8046aee:       4856            ldr     r0, [pc, #344]  ; (0x8046c48)
 8046afc:       4852            ldr     r0, [pc, #328]  ; (0x8046c48)
 8046b18:       484b            ldr     r0, [pc, #300]  ; (0x8046c48)
 8046b32:       4845            ldr     r0, [pc, #276]  ; (0x8046c48)
 8046b4c:       483e            ldr     r0, [pc, #248]  ; (0x8046c48)
 8046b68:       4837            ldr     r0, [pc, #220]  ; (0x8046c48)
 8046b76:       4834            ldr     r0, [pc, #208]  ; (0x8046c48)
 8046b94:       482c            ldr     r0, [pc, #176]  ; (0x8046c48)
 8046ba2:       4829            ldr     r0, [pc, #164]  ; (0x8046c48)
 8046bbe:       4822            ldr     r0, [pc, #136]  ; (0x8046c48)
 8046bcc:       481e            ldr     r0, [pc, #120]  ; (0x8046c48)
 8046bda:       481b            ldr     r0, [pc, #108]  ; (0x8046c48)
 8046c02:       4811            ldr     r0, [pc, #68]   ; (0x8046c48)
 8046c10:       480d            ldr     r0, [pc, #52]   ; (0x8046c48)
 8046c1e:       480a            ldr     r0, [pc, #40]   ; (0x8046c48)
 8046c38:       4803            ldr     r0, [pc, #12]   ; (0x8046c48)

next week, i'll look on the calling functions .. to see with task is involved 
and i'll look to find the I2C address   

 on https://github.com/travisgoodspeed/md380tools/issues/40 
-->
  Software Control Watch Dog Serial Clock
  Software Control Watch Dog Serial Data

```

aeickho, 2016-03-04:
```
af+ 0x08046a28 306 Big_I2C_Function()   
af+ 0x0804ff74 244 Function_Called_Big_I2C_Function()
af+ 0x08050068 270 Function_Function_Called_Big_I2C_Function()
af+ 0x08046e00 156 Function_Function_Function_Called_Big_I2C_Function()
af+ 0x08032a3c 146 Function_Function_Function_Function_Called_Big_I2C_Function(
CCa 0x08032a3c Called from Task Start()

```

tristantech, 2016-05-04:
I became curious about this too. It has something to do with the vocoder. The IC is called the HR_V3000S according to the schematic. There is very little information out there about it, sadly. I found lots of [other Chinese DMR radios](http://fjlisheng.en.alibaba.com/product/60182522239-221438579/DMR_Tier_2_Mototrbo_System_DM_R77_DMR_System.html) mentioning use of the  "DVIS HRV3000" as the vocoder. I presume "DVIS" is a knock-off of DVSI, which makes the real AMBE vocoders. "HR" is the same prefix as the DMR processor IC. It would not surprise me that China has reverse engineered or implemented the AMBE+ spec. However, to say that this 6-pin package, connected soley over I2C, _is_ the vocoder may be a big stretch. Perhaps it's a hardware token/dongle to activate the actual vocoder, inside the STM32?

Tapping the I2C bus would provide some valuable information. There is also a pack of code called "HR_V3000-vocoder-use" floating around on a bunch of sketchy Chinese download sites that you can find through Google. They make you either pay or upload lots of stuff to get the files. I think it would be very valuable to get a look at whatever that code is.

Tristan Honscheid / KD8YHB

aeickho, 2016-05-20:
from the dr6000 schematic ...  alpu-mp -> 
http://wenku.baidu.com/view/cb189e29dd36a32d737581aa.html?re=view
http://www.xcjy876.com/html/98y56bz6aa0370za40425y77.html

kb9mwr, 2016-06-13:
How to download from wenku.baidu: http://pastebin.com/ynJ9kLdU

Else: http://www.qsl.net/kb9mwr/projects/dv/dmr/ALPU-MP_datasheet_v2.16.pdf

aeickho, 2016-06-13:
Tnx

aeickho, 2016-07-14:
now on #168

closed 2016-07-14 by aeickho