travisgoodspeed/md380tools #49: 400-480 to 136-174 conversion

bilbolodz, 2016-03-22:
Does anyone know is it possible to reprogram band in radio? Is it firmware or hardware issue?

smiffys, 2016-03-22:
Its hardware you cannot change the band with firmware

bilbolodz, 2016-03-22:
How about firmware? Is it "universal" (UHF/VHF)? I would like to buy 136-174 version but also I want to have chance to load moded firmware a play with these new firmware.

ajorg, 2016-03-22:
It's hardware.

cr, 2016-03-22:
The UHF and VHF hardware uses identical firmware in the MD380. I would expect it to just work, but be advised that we have no feedback that it actually does, yet.

smiffys, 2016-03-22:
It does work fine, I have patched 2.032 firmware in my Retevis RT3 VHF and UHF radios

travisgoodspeed, 2016-03-22:
I'm also using it on the VHF model with no problems. Closing issue.

closed 2016-03-22 by travisgoodspeed

ajorg, 2016-03-22:
Oh, I get it. Sorry for chiming in without understanding first.