travisgoodspeed/md380tools #50: Motorola Basic Privacy compatibility.

travisgoodspeed, 2016-03-26:
Motorola's Basic Privacy can be made compatible with the MD380 by hooking `aes_cipher()` to return a motorola keystream.  Close this issue when we can make (poorly) encrypted calls between Motorola and Tytera radios.

travisgoodspeed, 2016-03-26:
Pushed the first AES hook in c18f6dde316db68be33fda65fa67593585bd42a9.  This prints first the Enhanced Privacy key, and then the keystream.  Only the first 59 bits of the keystream are used.

bilbolodz, 2016-03-27:
Current patch produce during compile a lot of warnings.

src/menu.c:34:15: warning: assignment makes pointer from integer without a cast
     menustruct=*((int*)menu + 2);
[...]
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /tmp/ccOA0K43.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail

travisgoodspeed, 2016-03-27:
The menu warnings are easy to clean up, but the wchar_t warnings are more complicated.  We need to compile to use two-byte wchar_t values, as that's why Tytera uses in their firmware, but the default compiler toolchains for ARM seem to all use four bytes.

We're being warned because while our code uses 2-byte wchars, the toolchain's libc still expects 4-byte wchars.  Rather than warn us when the functions cross, it expects us to rebuild the whole toolchain, but that would cause trouble between those of us who don't enjoy compiler toolchains from scratch and whatever sick bastards do enjoy it.

There's no good way to clean that up without requiring folks to rebuild their toolchain, so for now, we're sort of stuck with it.  Sorry.

travisgoodspeed, 2016-05-25:
Now that I have some Motorola radios, I'll finish this off.  Expect updates in the next week or two.

countryboy14057, 2016-06-24:
Any update if the MD380 will decode Basic Privacy ?

travisgoodspeed, 2016-06-24:
It will, but only after I spend a weekend coding it.  All hardware is available and all protocols are understood, just a matter of implementing the code.

travisgoodspeed, 2016-06-26:
This is an SMS of the word "Test" from 3147092 to 99.  The first recording is encrypted with Basic Privacy Key 1, while the second is cleartext.  The encrypted data bytes are predictably the same for this key.

```
SMS header:  08 6a 82 9a 00 00 63 30 05 54 86 00 18 3a
SMS header:  08 62 4f 10 01 00 00 00 00 00 00 00 03 b5
Data:        08 7a 5a 00 1f 2e 41 37 1f 00 1e 11 77 a1
Data:        08 72 13 30 1a 54 fe 00 1f 63 10 a7 10 a7
Data:        08 7a 1f 1a 1d 6e 1f 10 bf 00 8b 04 12 00
Data:        08 7a 15 00 4b 00 7a 00 6c 00 6b 00 00 00
Data:        08 72 00 00 00 00 00 00 00 00 ee 70 d0 53

SMS header:  08 62 82 4a 00 00 63 30 05 54 85 00 fe fc
Data:        08 7a 45 00 00 2e 5e 38 00 00 01 11 68 a0
Data:        08 72 0c 30 05 54 e1 00 00 63 0f a7 0f a7
Data:        08 72 00 1a 01 6e 00 10 a0 00 95 04 0d 00
Data:        08 7a 0a 00 54 00 65 00 73 00 74 00 00 00 -- 54657374=Test
Data:        08 7a 00 00 00 00 00 00 00 00 ef c9 b6 00
```

Here is the same SMS encrypted with Basic Privacy Key 2.  Note that the header is identical to Key 1, but that the body differs.  Also note that the null bytes retain their places, and aren't scrambled like the test bytes.
```
SMS header:  08 62 82 9a 00 00 63 30 05 54 86 00 18 3a
SMS header:  08 6a 4f 10 01 00 00 00 00 00 00 00 03 b5
Data:        08 72 a6 00 e3 2e 45 c2 e3 00 e2 11 c3 16
Data:        08 72 ef 30 e6 54 02 00 e3 63 ec a7 ec a7
Data:        08 72 e3 1a 19 6d e3 10 43 00 7f 04 ee 00
Data:        08 72 e9 00 b7 00 86 00 90 00 97 00 00 00
Data:        08 7a 00 00 00 00 00 00 00 00 1c 74 77 dc
```

Here is "Hello hello" encrypted with BP1 from 3147092 to 99.  First clear, then encrypted.  Note how "Hello" becomes "Wzssp" and "hello" becomes "wzssp".

```
SMS header:  08 62 82 48 00 00 63 30 05 54 86 00 6d c8
Data:        08 7a 45 00 00 3c 54 1d 00 00 01 11 72 ad
Data:        08 7a 0c 30 05 54 e1 00 00 63 0f a7 0f a7
Data:        08 7a 00 28 70 41 00 1e a0 00 9e 04 0d 00
Data:        08 7a 0a 00 48 00 65 00 6c 00 6c 00 6f 00 -- 48656c6c6f=Hello
Data:        08 72 20 00 68 00 65 00 6c 00 6c 00 6f 00 -- 68656c6c6f=hello
Data:        08 7a 00 00 00 00 00 00 00 00 62 63 5f bf

SMS header:  08 6a 82 98 00 00 63 30 05 54 87 00 ed 6c
SMS header:  08 6a 4f 10 01 00 00 00 00 00 00 00 03 b5
Data:        08 7a 5a 00 1f 3c 4b 1e 1f 00 1e 11 6d ac
Data:        08 72 13 30 1a 54 fe 00 1f 63 10 a7 10 a7
Data:        08 72 1f 28 70 41 1f 1e bf 00 80 04 12 00
Data:        08 72 15 00 57 00 7a 00 73 00 73 00 70 00 -- 577a737370=Wzssp
Data:        08 7a 3f 00 77 00 7a 00 73 00 73 00 70 00 -- 777a737370=wzssp
Data:        08 72 00 00 00 00 00 00 00 00 19 a5 08 09
```

travisgoodspeed, 2016-06-26:
So text encryption for BP1 is XOR of the text with 0x1F, and encryption with BP2 is XOR with 0xE3.  These bytes are conveniently exposed by XOR with a null byte on the third Data frame, so we don't need a lookup table to decrypt them.

travisgoodspeed, 2016-06-26:
Given a key of 179 (0xB3), we get this pair for the "Hello hello" message:
```
SMS header:  08 62 82 48 00 00 63 30 05 54 86 00 6d c8
Data:        08 72 45 00 00 3c 7a 57 00 00 01 11 4c 73
Data:        08 7a 0c 30 05 54 e1 00 00 63 0f a7 0f a7
Data:        08 72 00 28 7d 41 00 1e a0 00 91 04 0d 00
Data:        08 7a 0a 00 48 00 65 00 6c 00 6c 00 6f 00
Data:        08 7a 20 00 68 00 65 00 6c 00 6c 00 6f 00
Data:        08 7a 00 00 00 00 00 00 00 00 86 ce 2d c2

SMS header:  08 6a 82 98 00 00 63 30 05 54 87 00 ed 6c
SMS header:  08 6a 4f 10 01 00 00 00 00 00 00 00 03 b5
Data:        08 72 99 b9 dc 85 a6 e1 dc b9 dd a8 90 cb
Data:        08 7a d0 89 d9 ed 3d b9 dc da d3 1e d3 1e
Data:        08 72 dc 91 a0 f8 dc a7 7c b9 4e bd d1 b9
Data:        08 72 d6 b9 94 b9 b9 b9 b0 b9 b0 b9 b3 b9
Data:        08 7a fc b9 b4 b9 b9 b9 b0 b9 b0 b9 b3 b9
Data:        08 72 00 00 00 00 00 00 00 00 1d 19 cd 95
```

Unlike B1 and B2, B179's XOR mask fills both bytes.  The easiest place to find the mask is from the first line of Data, where `00 00` is XORed to become `b9 dc`.

janipewter, 2016-07-21:
Any update on this?

travisgoodspeed, 2016-07-21:
It's a low priority, but I'll get around to it soon enough.

janipewter, 2016-07-22:
Great. It would actually be incredible!

cryptonomicon, 2016-07-22:
IDK if this will help you make moto BP work with Tytera EP, but here are some of the outputs from Moto BP keys. Maybe you can see if you can match this to a Tytera:

Moto BP key# / 16 bit key/ First BP voice frame:

1/ 1F00 / e701b69f93e080
2/ E300 / 1b014a9f6fe000
3/ FC00 / 0401559f70e000
4/ 2503 / dd028c9ca9e380
5/ 3A03 / c202939cb6e380
6/ C603 / 3e026f9c4ae300
7/ D903 / 2102709c55e300
8/ 4A05 / b204e39ac6e580
9/ 5505 / ad04fc9ad9e580
10/ A905 / 5104009a25e500

hamradioPRC, 2016-07-24:
Is there some possibility you can write it to scan through all 255 BP keys?

cryptonomicon, 2016-07-26:
11- b605
12- 6f06
13- 7006
14- 8c06
15- 9306
16- 2618
17- 3918
18- c518
19- da18
20- 031b
21- 1c1b
22- e018
23- ff1b
24- 6c1d
25- 731d
26- 8f1d
27- 901d
28- 491e
29- 561e
30- aa1e

No need to scan through all 255 privacy codes.
It's a very simple system that is a low level privacy scheme. I tested these above keys with the results of my XPR 6550 and confirm them.

Did they ever release the AMBE2 USB frame dump code for the 380? You might be able to test that way?
Not sure what's available on a hobby level but you have to look at the data or voice frames.

I took a look at the 380 a bit ago and as far as compatibility for this my results were mixed/below average on intelligibility.

countryboy14057, 2016-07-27:
I'd be willing to offer a donation to get this working since some places around here are using it. Or even better I'd offer a donation to have someone add BP decryption into DSDPlus

hamradioPRC, 2016-07-27:
Dsd plus won't release their 'secret' version with decryption  as they know of vulnerability to businesses using them

countryboy14057, 2016-07-28:
No but I am sure someone out there can add it in pretty quickly. I wish I had gone for CS instead of System Admin somedays.

d235j, 2016-07-28:
DSPlus is a closed-source fork of DSD and only the maintainers of it can add functionality.

countryboy14057, 2016-07-28:
is DSD open sourced tho ?

hamradioPRC, 2016-07-31:
You will have to ask the DSD Fast Lane team

d235j, 2016-08-01:
DSD is open source, DSDPlus is not. I'm rather disappointed that DSDPlus took DSD and made it a closed-source proprietary product, but there's not much I can do about that.

hamradioPRC, 2016-08-03:
Any new updates to having BP key compatibility?

g738l, 2016-09-06:
I would like to ask if anyone would put Moto BP key 77 in the format as "Cryptonomicon"
thank you very much

eb4fbz, 2016-09-06:
Sorry we don't have all Moto BP keystreams, i only extracted 37 keys. It's a tedious work, it would be better if someone with crypto/maths knowledge could try to reverse engineer the keystream generation algorithm (supposed to be an LFSR).

g738l, 2016-09-06:
Thanks for the answer, I'll have to buy a Motorola  :-)

hamradioPRC, 2016-09-06:
Has keystream 10 been coded into MD380Tools yet please?

SatSearching, 2016-09-08:
Any chance some work can be done on this please?

Not wanting a blatant hack solution, but just compatibility for BP on the md380 to Mototrbo BP would be a very cool feature.

Moto117, 2016-09-09:
Hey guys... .thought I would throw help you out a little with this, would be nice to see BP working in the 380.

1	1F00
2	E300
3	FC00
4	2503
5	3A03
6	C603
7	D903
8	4A05
9	5505
10	A905
11	B605
12	6F06
13	7006
14	8C06
15	9306
16	2618
17	3918
18	C518
19	DA18
20	031B
21	1C1B
22	E01B
23	FF1B
24	6C1D
25	731D
26	8F1D
27	901D
28	491E
29	561E
30	AA1E
31	B51E
32	4B28
33	5428
34	A828
35	B728
36	6E2B
37	712B
38	8D2B
39	922B
40	012D
41	1E2D
42	E22D
43	FD2D
44	242E
45	3B2E
46	C72E
47	D82E
48	6D30
49	7230
50	8E30
51	9130
52	4833
53	5733
54	AB33
55	B433
56	2735
57	3835
58	C435
59	DB35
60	0236
61	1D36
62	E136
63	FE36
64	2B49
65	3449
66	C849
67	D749
68	0E4A
69	114A
70	ED4A
71	F24A
72	614C
73	AE4C
74	824C
75	9D4C
76	444F
77	5B4F
78	A74F
79	B84F
80	0D51
81	1251
82	EE51
83	F151
84	2852
85	3752
86	CB52
87	D452
88	4754
89	5854
90	A454
91	BB54
92	6257
93	7D57
94	8157
95	9E57
96	6061
97	7F61
98	8361
99	9C61
100	4562
101	5A62
102	A662
103	B962
104	2A64
105	3564
106	C964
107	D664
108	0F67
109	1067
110	EC67
111	F367
112	4679
113	5979
114	A579
115	BA79
116	637A
117	7C7A
118	807A
119	9F7A
120	0C7C
121	137C
122	EF7C
123	F07C
124	297F
125	367F
126	CA7F
127	D57F
128	4D89
129	5289
130	AE89
131	B189
132	688A
133	778A
134	8B8A
135	948A
136	078C
137	188C
138	E48C
139	FB8C
140	228F
141	3D8F
142	C18F
143	DE8F
144	6B91
145	7491
146	8891
147	9791
148	4E92
149	5192
150	AD92
151	B292
152	2194
153	3E94
154	C294
155	DD94
156	0497
157	1B97
158	E797
159	F897
160	06A1
161	19A1
162	E5A1
163	FAA1
164	23A2
165	3CA2
166	C0A2
167	DFA2
168	4CA4
169	53A4
170	AFA4
171	B0A4
172	69A7
173	76A7
174	8AA7
175	95A7
176	20B9
177	3FB9
178	C3B9
179	DCB9
180	05BA
181	1ABA
182	E6BA
183	F9BA
184	6ABC
185	75BC
186	89BC
187	96BC
188	4FBF
189	50BF
190	ACBF
191	B3BF
192	66C0
193	79C0
194	85C0
195	9AC0
196	43C3
197	5CC3
198	A0C3
199	BFC3
200	2CC5
201	33C5
202	CFC5
203	D0C5
204	09C6
205	16C6
206	EAC6
207	F5C6
208	84D0
209	85DF
210	8AD3
211	8BDC
212	B6D5
213	B7DA
214	B8D6
215	B9D9
216	D0DA
217	D1D5
218	DED9
219	DFD6
220	E2DF
221	E3D0
222	ECDC
223	EDD3
224	2DE8
225	32E8
226	CEE8
227	D1E8
228	08EB
229	17EB
230	EBEB
231	F4EB
232	67ED
233	78ED
234	84ED
235	9BED
236	42EE
237	5DEE
238	A1EE
239	BEEE
240	0BF0
241	14F0
242	E8F0
243	F7F0
244	2EF3
245	31F3
246	CDF3
247	D2F3
248	41F5
249	5EF5
250	A2F5
251	BDF5
252	64F6
253	7BF6
254	87F6
255	98F6

eb4fbz, 2016-09-09:
Great! Thanks a lot Moto117. So these are the 49bits keystreams:
1	1F 00 1F 00 1F 00 0
2	E3 00 E3 00 E3 00 1
3	FC 00 FC 00 FC 00 1
4	25 03 25 03 25 03 0
5	3A 03 3A 03 3A 03 0
6	C6 03 C6 03 C6 03 1
7	D9 03 D9 03 D9 03 1
8	4A 05 4A 05 4A 05 0
9	55 05 55 05 55 05 0
10	A9 05 A9 05 A9 05 1
11	B6 05 B6 05 B6 05 1
12	6F 06 6F 06 6F 06 0
13	70 06 70 06 70 06 0
14	8C 06 8C 06 8C 06 1
15	93 06 93 06 93 06 1
16	26 08 26 18 26 18 0
17	39 08 39 18 39 18 0
18	C5 08 C5 18 C5 18 1
19	DA 08 DA 18 DA 18 1
20	03 0B 03 1B 03 1B 0
21	1C 0B 1C 1B 1C 1B 0
22	E0 0B E0 1B E0 1B 1
23	FF 0B FF 1B FF 1B 1
24	6C 0D 6C 1D 6C 1D 0
25	73 0D 73 1D 73 1D 0
26	8F 0D 8F 1D 8F 1D 1
27	90 0D 90 1D 90 1D 1
28	49 0E 49 1E 49 1E 0
29	56 0E 56 1E 56 1E 0
30	AA 0E AA 1E AA 1E 1
31	B5 0E B5 1E B5 1E 1
32	4B 08 4B 28 4B 28 0
33	54 08 54 28 54 28 0
34	A8 08 A8 28 A8 28 1
35	B7 08 B7 28 B7 28 1
36	6E 0B 6E 2B 6E 2B 0
37	71 0B 71 2B 71 2B 0
38	8D 0B 8D 2B 8D 2B 1
39	92 0B 92 2B 92 2B 1
40	01 0D 01 2D 01 2D 0
41	1E 0D 1E 2D 1E 2D 0
42	E2 0D E2 2D E2 2D 1
43	FD 0D FD 2D FD 2D 1
44	24 0E 24 2E 24 2E 0
45	3B 0E 3B 2E 3B 2E 0
46	C7 0E C7 2E C7 2E 1
47	D8 0E D8 2E D8 2E 1
48	6D 00 6D 30 6D 30 0
49	72 00 72 30 72 30 0
50	8E 00 8E 30 8E 30 1
51	91 00 91 30 91 30 1
52	48 03 48 33 48 33 0
53	57 03 57 33 57 33 0
54	AB 03 AB 33 AB 33 1
55	B4 03 B4 33 B4 33 1
56	27 05 27 35 27 35 0
57	38 05 38 35 38 35 0
58	C4 05 C4 35 C4 35 1
59	DB 05 DB 35 DB 35 1
60	02 06 02 36 02 36 0
61	1D 06 1D 36 1D 36 0
62	E1 06 E1 36 E1 36 1
63	FE 06 FE 36 FE 36 1
64	2B 09 2B 49 2B 49 0
65	34 09 34 49 34 49 0
66	C8 09 C8 49 C8 49 1
67	D7 09 D7 49 D7 49 1
68	0E 0A 0E 4A 0E 4A 0
69	11 0A 11 4A 11 4A 0
[... 163 lines cut ...]
233	78 0D 78 ED 78 ED 0
234	84 0D 84 ED 84 ED 1
235	9B 0D 9B ED 9B ED 1
236	42 0E 42 EE 42 EE 0
237	5D 0E 5D EE 5D EE 0
238	A1 0E A1 EE A1 EE 1
239	BE 0E BE EE BE EE 1
240	0B 00 0B F0 0B F0 0
241	14 00 14 F0 14 F0 0
242	E8 00 E8 F0 E8 F0 1
243	F7 00 F7 F0 F7 F0 1
244	2E 03 2E F3 2E F3 0
245	31 03 31 F3 31 F3 0
246	CD 03 CD F3 CD F3 1
247	D2 03 D2 F3 D2 F3 1
248	41 05 41 F5 41 F5 0
249	5E 05 5E F5 5E F5 0
250	A2 05 A2 F5 A2 F5 1
251	BD 05 BD F5 BD F5 1
252	64 06 64 F6 64 F6 0
253	7B 06 7B F6 7B F6 0
254	87 06 87 F6 87 F6 1
255	98 06 98 F6 98 F6 1

It would be great if we could run the Tytera AES function to search for matches, they should be plenty. That way the BP compatibility would not need any patch and could be used with stock firmware too.

cryptonomicon, 2016-09-09:
DMR plaintext voice frame is:
F8 01 A9 9F 8C E0 80

Apply BP mask to pt and compare that to what the MD 380 EP outputs and look for a match.

eb4fbz, 2016-09-09:
I have an idea: why not patch the Tytera basic privacy function so it behaves as Moto BP?

Tytera basic privacy is not recommended, it's a xor over the full voice frames (including FEC), resulting in high understandability with some keys in the clear or with "near" keys (because FEC helps it), and the worst of all: scrambling over FEC produces errors when used through repeaters, reducing range.

hamradioPRC, 2016-09-20:
Converting A90A from 16bit hex is 10 in decimal (8bit)

Thought I might throw that one in somewhere

SatSearching, 2016-09-23:
Great that some have posted the 256 privacy codes to help, but is there any plans to work on compatibility of BP for MD380 please?

travisgoodspeed, 2016-09-23:
I'll be writing it when I find the time, but time is short and it's a low priority. 

73 from Pizza Rat City.

SatSearching, 2016-09-24:
Thanks for the reply, I appreciate what you mean about having the time though, I am no coder but done some for embedded satellite TV boxes. I wish I could help with this project but python is not my thing.

However, can I convince you that BP compatibility should be high priority and not low? I would love to see this feature, and sure many others would love to see.
http://s2.quickmeme.com/img/c5/c5d5b517c9a526eb953dd4fd121667040b3512959c8de616ec8f2f01190d3540.jpg

73 from Yorkshire in the UK

hamradioPRC, 2016-09-24:
I agree. This should be set as a higher priority, as users are more eager to listen in.

travisgoodspeed, 2016-09-24:
Where is encrypted DMR found in the wild?  All commercial repeaters that I've found seem to be cleartext.

If you can give me a frequency in Philly, Knoxville, or NYC, I'll be a lot more motivated to finish this feature.

73 from Philly

janipewter, 2016-09-24:
Maybe not so much in the states where DMR is not widely used on commercial syetems, but privacy is common in the UK. I would rate this feature as probably the highest importance right now

Forts117, 2016-09-25:
And they are using basic, not enhanced? Why anyone would use basic is beyond me... Short of just wanting to block DMR scanners.

cryptonomicon, 2016-09-25:
The 40 bit EP isn't secure either.

It serves the same purpose as BP. It stops whackers.

BP and RAS is generally good enough.

Forts117, 2016-09-25:
No, but it's a whole lot more secure than basic is. On a busy system you can find a BP value by trial and error in no time. With EP you still have over a trillion keys to search.

eb4fbz, 2016-09-25:
About half the encripted DMR networks i found use EP, and some RAS (RAS doesn't prevent scanners). BP security level is a joke, but up to date there are no download&run tools for it, so it stops most whackers.

I would not support such a tool because the only consequence would be that all BP networks would switch to EP. It's a free option, although using EP reduces the range a little. I wouldn't consider reversing BP a code breaking (which is ilegal here) because it's a simple math operation against a known published text, but EP would be because of the RC4 algorithm. 
And even if an EP cracking tool is published, Moto would then release AES option for free. It happened here with Tetra, and now almost all networks use authentication because they were suffering congestion because of the terminals from eavesdropers being actively registered and using slots irresponsibly.

So i think this is senseless for most of us. My only motivation for pushing BP in MD380tools is for interoperability with legitimate Moto networks. 
If your intention is to eavesdrop BP or EP, better use dsd160 (open source), implement it yourself with automatic key recovery, and please keep it private, be responsible.

SatSearching, 2016-09-25:
I am not suggesting a hack/crack, just suggesting making the MD380 compatible with Motorola BP

eb4fbz, 2016-09-25:
Some do, i just wanted to clarify my position.

SatSearching, 2016-09-26:
So, can we agree that BP compatibility should be a higher priority than it currently is?

Not a crack/hack solution, just compatibility I would like to see, It sounds the issue is where the EXOR is happening with the MD380 and BP....?

cryptonomicon, 2016-09-26:
The issue is the way the keystream is generated.

I've tried a few things between an md380 and a XPR 6550 regarding EP/BP compatibility and it is not perfect but you could make out some of the conversation.

As someone said BP is a joke, but so is 40 bit EP.
No one is going to try key after key by hand.
BP can be had instantly already (there exists a receiver) and most EP systems are set up by non-professionals and a hypothetical attacker could thus recover a key in seconds to a couple hours due to the common 110x reduction in key space.

Not looking to hack either. It would be great if RC4 could be dropped in a 380 too with front panel key entry (known key) or cps.

I just try to advise people neither forms of voice privacy (BP and 40 bit EP) are secure.

I can run test vectors on BP and RC4 if needed.
Not to hack, but to work on getting these two modes working in an MD380.

I don't know if moto would be happy about this though. ARC4 is open but moto BP is Motorola exclusive and if they have any trade secrets, patents ,etc I'm avoiding it.

Then we could have full interop with other moto radios in privacy mode.

If everything is legal of course.

janipewter, 2016-09-27:
I'm not interested in a HACK. I would just like compatibility so I can use my MD380 on encrypted Motorola systems that I know the privacy key for.

cryptonomicon, 2016-09-27:
Then you should not participate with experimental firmware for the 380 because that is a "hack".

Getting things to do something they weren't designed to do is considered a hack.

So you're all hackers by definition.

sijskes, 2016-09-27:
@SatSearching imho there is no such thing as prioritizing in this project. It is the same as other wishlist items, you have to do the work yourself if you want it now.

cryptonomicon, 2016-09-29:
BP 77/ 5B4F/ A3 0E F2 D0 D7 AF 80

Kpt- F8 01 A9 9F 8C E0 80
xor
Ct- A3 0E F2 D0 D7 AF 80
= Ks 5B 0F 5B 4F 5B 4F 00

cryptonomicon, 2016-09-29:
That was for you g738I.

Hope it helps with future BP interop with the MD380.

g738l, 2016-09-29:
Thank you very much Mister cryptonomicon.

eb4fbz, 2016-10-04:
Motorola has no published patent about BP. I really doubt that could be patented, and even if it could, having it's details published in the patent would be stupid.

hamradioPRC, 2016-10-13:
Not seen any progress being made so far. What's happening?

travisgoodspeed, 2016-10-13:
No new progress, but pull requests are always welcome.

hamradioPRC, 2016-10-16:
Can EP be workable by changing 128bit chars down to 40bit chars? Reason for asking is there seems to be  systems in my area that are using a lower bit layer encryption and would like to see this worked comparable with a motorola

Forts117, 2016-10-16:
It's completely different algorithms. Hytera 128 is 'supposedly' AES128 whereas EP is 40 bit RC4.

cryptonomicon, 2016-10-16:
You guys are confusing privacy schemes.

The guy wanted to know if Tytera 128 bit privacy can be reduced to 40 bit encryption to work with MotoTRBO 40 bit encryption.

The two schemes are different. Tytera 128 bit privacy is static. By that I mean it uses the same keystream to xor with the clear text.

MotoTRBO 40 bit encryption uses RC4.
The secret key is concatenated with a 4 byte MI that changes with each superframe.

With 40 bit keys, a changing IV adds no extra security.

MotoTRBO EP/ Hytera 40 bit Full Encrypt:
Enc key+ IV (MI)= key stream
Keystream is XOR'd with each superframe which forms the cipher text.

DMR MI Generation:
4 byte input and a 32 stage shift using the following polynomial- x^32+x^4+x^2+1 per ETSI spec.

Hypothetical:
Key:0000000001
IV: c609ab01
Voice frame 1 would be: ff60899e469485
Xor pt and first Enc frame to yield the rc4 key stream (*not the actual key!)
Next IV would be: b14efa29, then 78febe5a, 4ffeea65, 4925c7e3.....

My conclusion, rc4 is easy to implement but I don't see much work continuing on the MD 380.

Get yourself a used xpr 6550 and buy the cps on eBay if you're willing.

Good luck.

hamradioPRC, 2016-10-22:
I have been busying myself making a raw sample of BP/EP off an encrypted motorola system. I can share files which are in both txt and wav. The txt files have been collected using DMR DECODE software and has binary output

[59 more replies]

closed 2018-09-27 by travisgoodspeed