travisgoodspeed/md380tools #53: fC instead of CC

Maijin, 2016-04-05:
Feel free to try `fC` instead of `CC` when using @radare :  https://github.com/travisgoodspeed/md380tools/blob/master/annotations/2.032/reg.r

aeickho, 2016-04-05:
tnx... i'm really an radare2 starter

Maijin, 2016-04-05:
everyone is somehow :new_moon_with_face:

aeickho, 2016-04-05:
@Maijin, I do not understand the use of fC.

Line 1 of reg.r:

CCa 0x800df8a "40020c00" "GPIOD GPIO port mode register (GPIOx_MODER)"

`0x0800df8a      dff8680b       ldr.w r0, [pc, 0xb68]       ; [0x800eaf4:4]=0x40020c00 ; "40020c00" "GPIOD GPIO port mode register (GPIOx_MODER)"`

"0x40020c00" is the address of "GPIOD GPIO port mode register (GPIOx_MODER)" ... yes I would like that all places where the accessed register 0x40020c00 comment appears that it is "GPIOD GPIO port mode register (GPIOx_MODER ) "is.

what is the right usage from fC in this case?

radare, 2016-04-05:
fC associates a comment to a flag, this way, the comments will appear everytime an instruction references the address pointed by the flag. Afaik there are some tests in the regressions suite for this. This was a feature request from @xvilka so may he can also provide some real life examples for this

XVilka, 2016-04-05:
set a corresponding flag name.e.g. GPIO1_MODER and then "GPIO port mode register" as fC name.

aeickho, 2016-04-05:
@XVilka can you write me some example? i have problems to understand your comment.

... fC "GPIO port mode register" GPIO1_MODER
but how to added 0x40020c00 to this entry ?

XVilka, 2016-04-05:
`f GPIO1_MODER 4 @ 0x40020c00`
`fC comment GPIO1_MODER`

radare, 2016-04-05:
Quotes dont work like this in r2land

aeickho, 2016-04-09:
ok, I'm understand the f command 
`f WelcomeLine1 4 @ 0x2001cecc
`

```
0x0804457c      dff80c0d       ldr.w r0, [pc, 0xd0c]       ; [0x804528c:4]=0x2001cecc WelcomeLine1                             
0x08044580      c9f783f9       bl gfx_drawtext             ;[6]                                                                
   
```
but i can't see the comment in  visual mode ...
`
fC WelcomeLine1 Welcome_Line1 str pointer
 `

also: is it possible that’s radare2 is generating an list with all places "WelcomeLine1" is used to me?

aeickho, 2016-04-27:
radare2 is complicated.  I learn when using radare2.

closed 2016-04-27 by aeickho