travisgoodspeed/md380tools #54: DSTAR/Fusion possibility

bilbolodz, 2016-04-08:
I've read today at Yahoo MD-380 statementL

> The hardware of the MD380 will not support anything other than DMR and Analog. You cannot write > firmware to do the other formats due to the limitation of the existing hardware.

Could someone comment this? Is it true? I would be a very sad message.

smiffys, 2016-04-08:
The C5000 chip is for DMR baseband but I would have thought it may be possible to bypass it and decode in software, personally I doubt it will ever happen, we can't even open a DMR channel without knowing the color code at the moment.

bilbolodz, 2016-04-08:
OK, thanks for info.

closed 2016-04-08 by bilbolodz

ajorg, 2016-04-08:
I definitely wouldn't say it's impossible, but it will be a piece of work, and it probably won't happen as an add-on to the existing firmware. I'd say the most likely outcome is that some more capable hardware comes out before any firmware for this hardware can do Fusion or D-STAR.

der-stefan, 2016-05-11:
A big problem, why we can't tell you for sure, is the lack of documentation. A complete developer manual of the C5000 could show us, what is possible. Is there a firmware running on the C5000, is the coding, modulation, demodulation, error correction etc. done in software or is it a pure ASIC? There is no chance to predict something without a proper documentation.
There are also voices that say that the schematic (see Github) is not 100% correct.