travisgoodspeed/md380tools #58: promiscuous mode switchable

aeickho, 2016-04-19:
to new menu items for {en,dis}able 

- [ ] talk groups 
- [ ] private calls

smiffys, 2016-04-19:
Great work, any joy with promiscuous color code or timeslot yet?

travisgoodspeed, 2016-04-19:
You can use 'md380-tool findcc` to identify the color code of a transmitter, but the C5000 chip hasn't got a setting for color code or timeslot promiscuity that I can find.

Now that @aeickho is doing extra menu items, it shouldn't be hard to move the color code finder into firmware and away from the host.

aeickho, 2016-04-19:
I'll try to write to the screen  continuously ...

travisgoodspeed, 2016-04-24:
Promiscuous mode is now done in C rather than assembler, and it is separately configurable for Talkgroups and Private Calls.  I've only added a menu item for TG promiscuity, but adding a Private option ought to be straightforward.

dysinger, 2016-04-24:
I was just thinking that I wanted this feature last night.  I woke up & merged it to my master.  Didn't even have to file an issue. :D Thanks!

rct, 2016-04-25:
FWIW, I created issue #61 to attempt to define and track a definition of basic DMR sniffing functionality that would be useful to a user in the field (without an attached computer).

Note: In addition to being able to hear what is being transmitted, it would be helpful if the user was able to determine what color code and time slot/talkgroup pair was being used.

travisgoodspeed, 2016-07-08:
Closing this without separating the menu items.

closed 2016-07-08 by travisgoodspeed