travisgoodspeed/md380tools #60: Menu operations should take precedence over receiving.

rct, 2016-04-25:
Using the stock firmware, the menu functions aren't accessible while the radio is receiving. Menu operations will be aborted if the radio starts to receive while in the menus. It should be possible to enter the menus and change settings while on a busy channel.

Close this when it is possible to change zones while receiving.

Ideally menu operations should possible while receiving a signal and decoding audio. However, it would be preferable if it was possible to activate the menus and abort the receive operation allowing the user to regain control over the radio.

(No idea if this is remotely feasible to fix via patching, bu.)

rct, 2016-04-25:
Note: some users consider this to be one of the radios top usability issues.  This should get the enhancement label.

travisgoodspeed, 2016-04-25:
This is a big issue, particularly when trying to send a short text during a busy conversation.

aeickho, 2016-05-04:
I'm work on it, but temporary I can't locate the technology. The message is not transported  by OSMBox :( ...

aeickho, 2016-05-06:
fixed with 352afdd83bdc341849bece388b26d4c978095186

I'm waiting for some response/ reports .. and an good config menu name

aeickho, 2016-05-06:
the functionality to enter the menu while receiving .. I think this is the same function ..

travisgoodspeed, 2016-05-08:
Good job!  This works perfectly, and it's incredibly useful.  Please enable it everywhere and without a configuration option. (We'll consider this a bug fix of the original firmware.)

Close this issue when the fix is enabled by default.

aeickho, 2016-05-08:
ok, but this patch is only the first part ... I'm work on a patch to enable the menu while rx.

is now enabled by default 
https://github.com/travisgoodspeed/md380tools/commit/9f005dc3235e68108a2957f86eeded97c6c100f0

aeickho, 2016-05-08:
not nice ... while rx and tx  @ 0x0801f044 will not be called

aeickho, 2016-05-08:
_"(We'll consider this a bug fix of the original firmware.)"_
maybe, I think ... more a code design problem

aeickho, 2016-05-08:
line of thought: if  (rx or  tx  aktive ) -> Status 0x11 the keyboard is configured for dtmf tones :(

aeickho, 2016-05-08:
update: keypad tones working while rx active

aeickho, 2016-05-08:
0x2001d26e  key code only while pressed

aeickho, 2016-05-09:
I'm stuck here

closed 2016-05-30 by aeickho