travisgoodspeed/md380tools #61: Basic DMR sniffing mode (standalone)

rct, 2016-04-25:
When a user finds a frequency with DMR activity, a standalone radio should be capable of identifying the color code, time slots and talk groups in use that would allow the user to monitor and capture the necessary information.

Close this when a standalone radio is capable of being tuned to a specific frequency to identify the color code(s) and some number of talk group/time slot pairs that have been recently used on that frequency.

(This is an attempt to define some basic sniffing functionality that would be helpful to a user in the field. Note: using the term "sniffing" for identifying the DMR parameters on a single frequency, as opposed to "scanning" which generally is associated with automatically checking multiple frequencies for activity.)

sijskes, 2016-10-02:
some data is already there, in netmon1:
* last voice call packet info
* last data header info
in "Alt. status":
* last used dst address in voicecall.

hamradioPRC, 2016-10-07:
Will netmon display last heard colour code of voice call as well as TG?

example: CC/TS/TG defined as CC:1 TS:2 TG:1

sijskes, 2016-10-07:
no. it displays the cc: ts1: ts2: flags from the radio 'layer' in netmon2

hamradioPRC, 2016-10-25:
@sijskes, I am currently playing around with FW build f62053f. Could it be possible if I could have Netmon 2 removed and DEBUG removed as I don't have any use for these? I am happy with using Netmon 1 and experimental modes. I am not a programmer so cant DYI. My email is on  my GitHub profile. Thanks.

hsmade, 2016-10-26:
@hamradioPRC You just gave your own answer. You are not a programmer, hence these modes are not developed for you. Please don't take offence in this, but you're basically asking to remove the diagnostics plug from your car here.

hamradioPRC, 2016-10-26:
OK but the question is NOT directed at you! It was for @sijskes. Surely he can do it using the source files, the code is already in place. Its just a matter of changing in that build.

roelandjansen, 2016-10-26:
if it's just a matter of changing, go ahead. The stuff is developer only and as is.

sijskes, 2016-10-26:
@hamradioPRC this discussion does not belong in this issue. please take it to google groups. this is not a forum.