travisgoodspeed/md380tools #62: Control or mitigate overly loud beep tones.

rct, 2016-04-25:
In the stock firmware, the beep tones emitted by the radio are at a fixed volume that ignores the setting of the volume knob. Some users are using a hardware mod as a work around.

Close this issue when it is possible to mitigate the beep tone volume in some manner.

(This is a top usability complaint for this radio/stock firmware.  Should also be on the feature list for any possible replacement firmware.)

Note: It looks like @aeickho has made progress on this in commit https://github.com/travisgoodspeed/md380tools/commit/5a9f3503b338d30f41c7f9b7564df55fb7e0eb39   Currently `README.me` does not list the features available in the experimental firmware.

aeickho, 2016-04-27:
@rct ,  you can hear the  difference between the two settings?

aeickho, 2016-04-27:
Nice Video 
https://www.youtube.com/watch?v=u9QCJNWrF2I

dg9vh, 2016-04-27:
Thanks for the tributes :-) It was a quick video without any preparation and without any script... I did my best - but had better wait for 3 days to show more nice features, hi.

closed 2016-05-08 by aeickho