travisgoodspeed/md380tools #63: Make error in Debian Jessie

Giorgiofox, 2016-04-26:
I've fresh installed a Debian Jessie, 
during Make I have this error:

../../md380-fw --unwrap ../../firmware/D002.032.bin unwrapped.img
DEBUG: reading "../../firmware/D002.032.bin"
Traceback (most recent call last):
  File "../../md380-fw", line 203, in <module>
    main()
  File "../../md380-fw", line 185, in main
    md.unwrap(input)
  File "../../md380-fw", line 114, in unwrap
    header = header.unpack(img[:256])
struct.error: unpack requires a string argument of length 256

No problem on Ubuntu, all library and requirement satisfied.

dl8bb, 2016-04-26:
Please verify that you realy have the file D002.032.bin in firmware folder and it has a length other than 0.

dl8bb@TARS:~/tmp/dl8bb/md380tools/firmware$ ls -ltr
total 3296
-rw-r--r-- 1 dl8bb dl8bb     441 Apr 25 12:45 Makefile
-rw-r--r-- 1 dl8bb dl8bb 2374886 Apr 25 16:04 D002.032.zip
-rw-r--r-- 1 dl8bb dl8bb  994816 Apr 25 16:04 D002.032.bin
dl8bb@TARS:~/tmp/dl8bb/md380tools/firmware$

travisgoodspeed, 2016-04-26:
If that is the trouble, we should add better checks before closing this issue.

aeickho, 2016-04-26:
And compare the md5sum  with annotation/2.032/flash.r line 15

Giorgiofox, 2016-04-26:
Solved, the file was different. Just re downloaded it, thanks to all.

travisgoodspeed, 2016-04-26:
Reopening.  Close this issue when the build scripts either (1) don't produce an empty file in response to network trouble or (2) identify the empty file and recreate it properly.

rct, 2016-04-26:
A reminder the Makefile depends on va3xpr's personal blog for the firmware files. 
```
D002.032.zip:
	curl 'http://www.va3xpr.net/?ddownload=9197' >D002.032.zip
D002.034.zip:
	curl 'http://www.va3xpr.net/?ddownload=9420' >D002.034.zip
```

Unfortunately the files on the Tytera site are .rar files with cryptic file names.
* http://www.tyt888.com/download.html
* http://www.tytera.com/download.html

ajorg, 2016-04-26:
It's also terribly slow to download from the manufacturer site.

> when the build scripts either (1) don't produce an empty file in response to network trouble

The `--fail` option of curl might be helpful. `curl --fail $url > $file || rm $file`

aeickho, 2016-04-27:
fixed in 142745eae777c9eb118cfa004bfe137afaf8a6e8

tnx to @dl8bb

closed 2016-04-27 by aeickho

iddq, 2016-04-28:
https://github.com/travisgoodspeed/md380tools/pull/75