travisgoodspeed/md380tools #66: VHF and CSV User List Compatibility

travisgoodspeed, 2016-04-26:
Giorgio reports that the VHF model of the MD380 doesn't like the addition of the CSV to the SPI Flash.  Somehow, it is damaging the codeplug.

[image]

Assigning to @aeickho, as the CSV database is his contribution.  I'll try reproducing this on my VHF MD380 tonight.

travisgoodspeed, 2016-04-26:
I've reproduced this bug, and the entire codeplug is definitely being obliterated during installation.  This was not caused by an incoming packet or frame collision.

```
iMac% md380-dfu read codeplug.one
...................................................Done.
Read complete
iMac% hexdump codeplug.one 
0000000 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
*
002f000 ff ff ff 01 ff ff ff ff ff ff ff ff ff ff ff ff
002f010 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
*
0040000
iMac% 
```

Looking a deeper, it seems that the VHF codeplug is much smaller, only 1MB instead of 16MB.  Thus there is no room for the DMR MARC database, and we ought to avoid flashing the database, as the erasure will take out the original image.

```
iMac% md380-tool spiflashdump vhf.whole
Dumping SPI Flash to 'vhf.whole'.
iMac% md380-tool spiflashdump uhf.whole
Dumping SPI Flash to 'uhf.whole'.
iMac% rabin2 -zzz vhf.whole | grep Marine\ 20A
0x0001fda0 Marine 20A Simp
0x0011fda0 Marine 20A Simp
0x0021fda0 Marine 20A Simp
0x0031fda0 Marine 20A Simp
0x0041fda0 Marine 20A Simp
0x0051fda0 Marine 20A Simp
0x0061fda0 Marine 20A Simp
0x0071fda0 Marine 20A Simp
0x0081fda0 Marine 20A Simp
0x0091fda0 Marine 20A Simp
0x00a1fda0 Marine 20A Simp
0x00b1fda0 Marine 20A Simp
0x00c1fda0 Marine 20A Simp
0x00d1fda0 Marine 20A Simp
0x00e1fda0 Marine 20A Simp
0x00f1fda0 Marine 20A Simp
iMac% rabin2 -zzz uhf.whole | grep Pizza
0x000149e0 Pizza Rat
iMac% 
```

aeickho, 2016-04-26:
sorry, check_spi_flash_type() is not included on  TDFU_SPIFLASHERASE64K: cmd 3...

aeickho, 2016-04-26:
fixed on b84e1d920ddc72e2a0e3818a7a23daac0a426b72

please repeat the spi write test and report the message of check_spi_flash_type() 

"no W25Q128FV %x %x %x..."

travisgoodspeed, 2016-04-26:
That seems to have fixed it.

Running the update for a little while, then abandoning it gives me this message.
```
iMac% md380-tool dmesg
22:11:37
Channel 1 selected.
Starting main()
aes_cipher(0x200125e4);
IN  : 06 f0 56 0e 22 00 2f 00 0e 51 35 30 36 38 33 36
OUT : d6 dc 8d 98 c2 30 09 46 9f 4e 96 bc 13 4d 28 fb
aes_cipher(0x200125e4);
IN  : 06 f0 56 0e 22 00 2f 00 0e 51 35 30 36 38 33 36
OUT : 14 b3 0e 5e 14 e7 2d f2 4d 59 3c 57 3a 70 fa f5
no W25Q128FV ef 40 14
no W25Q128FV ef 40 14
no W25Q128FV ef 40 14
no W25Q128FV ef 40 14
no W25Q128FV ef 40 14
no W25Q128FV ef 40 14
no W25Q128FV ef 40 14
no W25Q128FV ef 40 14
no W25Q128FV ef 40 14

iMac% 
```

We should add a USB command to query the SPI Flash's extra parameters over USB, as silently failing is a bit awkward to the user.

aeickho, 2016-04-27:
ef 40 14:

2.5V 8M-BIT
SERIAL FLASH MEMORY WITH
DUAL AND QUAD SPI

https://www.winbond.com/resource-files/da00-w25q80blf1.pdf

only 1MByte not 16MBytw

travisgoodspeed, 2016-05-22:
Given that this issue has been duplicated at #69, #78, #117, #128, and #131, it's probably time that we did something to properly close it.

I suggest that we either find a gap in the codeplug's lower 256K which is unused, or we compromise by putting the menu settings at 0xEBFC, where 72 bytes are available overlapping the very last two contacts.

Close this issue when menu settings are stored somewhere on VHF devices.

aeickho, 2016-05-22:
@travisgoodspeed please make this. I have no vhf radio actually.

travisgoodspeed, 2016-05-22:
No problem.  I'll patch it this week and close the issue.

commit 5b990062 (2016-05-24): Fixes the saving of menu settings on VHF radios.
#69, #78, #117, #128, #131.
Close #66.
closed 2016-05-24 by travisgoodspeed