travisgoodspeed/md380tools #69: Radios with small spi flash

aeickho, 2016-04-27:
regarding also #66

The problem is also relevant for  spi_flash_addl_config.

aeickho, 2016-04-27:
at the moment fixed with a1361860142d9569b907b5307d0c0760fa927c00

dg9vh, 2016-04-28:
Would in this case a solution to bring a userdb into the TRX for example with a reduced set (for example only contacts from 262xxxx or something)? Or is there no space left in memory for storing any further data?

aeickho, 2016-04-28:
see: 5dfb1afa202ddf73e0c0a4ac5faa3a5beb21148c  there is an version with the german db in code flash. Code Flash is to expensive: so i was very happy to find space in the codeplug flash.

btw. there is some possible "unused" space near the function to enter Chinese messages. "Chinese T9"

aeickho, 2016-05-18:
see #66

closed 2016-05-18 by aeickho