travisgoodspeed/md380tools #73: accents in users.csv

thorsen999, 2016-04-28:
Hi, would it be possible to replace the accents in user.csv, for example by re-encoding it in utf-8, I did it like that with a standalone python script but I'm just beginning with python so there must be a better way : 
import unicodedata
f = open('users.csv', 'r')
w = open('users_utf8.csv', 'w')
s = f.read()
s1 = unicode(s, 'utf-8', errors='ignore')
s2 = unicodedata.normalize('NFD', s1).encode('ascii', 'ignore')
w.write(s2)
w.close()

73 de Tom, F8COD

aeickho, 2016-04-28:
Important  Issue, see my note on #55 ...

travisgoodspeed, 2016-04-28:
Do accent marks exist in the UTF16 Latin font?  If so, we just need better UTF8->16 conversion.  If not, we'll have to either normalize to ASCII or add accent marks.

Maybe @cr or @maushammer have ideas.

sneakyness, 2016-05-01:
`export LC_CTYPE=C`
`export LANG=C`

made the sed bitching from `make flashdb` go away, but this might cause other bitching elsewhere in the future? 

[snack overflow question/answer rabbit hole](http://stackoverflow.com/questions/19242275/re-error-illegal-byte-sequence-on-mac-os-x)

maushammer, 2016-05-01:
Accents don't exist on the fonts in the firmware. 13,450 individual Chinese glyphs, and no accent marks on the Latin characters 😜   No Ø or other non-English characters, either.

There might be some software stuff to combine accents on top of characters, but I haven't looked at the code (and I don't have high hopes they did it this way).

travisgoodspeed, 2016-05-24:
My latest commit strips out the unicode characters.  We should probably do something smarter, such as converting unicode characters to their ASCII equivalents by dropping accent marks.

Be careful that further patches toward this issue do not pass unicode through `cut` or `sed`, which on OS X cannot handle it.  (`gcut` and `gsed` can, but of course they aren't available by those names under Linux.)

travisgoodspeed, 2016-05-25:
At @ajorg's suggestion in 3afedfad03ef11e1b9793e3b30c35a160527dbac, I tried `ICONV=iconv -f UTF-8 -t ascii//TRANSLIT` but transcription errors still exist.

The first culprit I find is F1HBG (2080002) who lives in Île-de-France, but whose encoding places him in `Ãle-de-Franc`.  In which encoding is this valid, and for how many encodings will we need to fix up DMR MARC's database?

Since TRANSLIT is correct for the majority of cases, I'm applying that patch.

commit 75444095 (2016-05-25): Now attempts to convert from Unicode, with a few errors from non-unicode formats.
Close #73.
closed 2016-05-25 by travisgoodspeed