travisgoodspeed/md380tools #77: Extra Top-Icons

travisgoodspeed, 2016-04-29:
We need the ability to show additional icons, indicating features such as promiscuous mode and a custom scanner.  These could go in the original row, or they might be shown along the right-hand side of the display.

aeickho, 2016-05-09:
@all .. this is an important feature for the hamvention release :) 

and in my opinion it is quite easy to realize a beautiful project to go into the software development on the md380.

sample image: 0x80f9d20.ppm 

have fun
alex

aeickho, 2016-05-09:
first step: create an image 12x12 pixel for the promiscuous mode

phl0, 2016-05-09:
What about this?
[image]

gadget-fan, 2016-05-09:
Or this.
[image]

maushammer, 2016-05-09:
I'll throw an illuminati-style pyramid eye in to the ring:
[image]

der-stefan, 2016-05-09:
Or this:
[image]

maushammer, 2016-05-09:
Or a magnifying glass "inspector":
[image]

gadget-fan, 2016-05-09:
I like the pyramid!  There is this too.
[image]

der-stefan, 2016-05-09:
Or just simply an eye:
[image]

phl0, 2016-05-11:
I am stuck trying to integrate this icon. Could someone assist me with integrating the icon into the firmware?

dg9vh, 2016-05-12:
@phl0 maybe i could help you. Where exactly is the problem? have you created a ppm-file? have you modified the head of the file (if necessary)?

phl0, 2016-05-15:
@dg9vh Jepp. Created a ppm file and changed the header to include the data of the original .ppm file. The question is how to import it back into the firmware.

dg9vh, 2016-05-15:
Should work like replace the welcome screen shown in makefile but with other address.

aeickho, 2016-07-09:
[image]

travisgoodspeed, 2016-07-09:
Nice!

aeickho, 2016-07-09:
done with f07304c2500983d044b1eb0ed2d99df90ece103e

the place to update the eye is not perfect ... but I have the location where battery, letterhead, etc. is drawn not found

aeickho, 2016-09-21:
the Top Icon line is full, on the position of the "eye" is also the gps (locked) icon

[image]

closed 2016-09-21 by aeickho