travisgoodspeed/md380tools #78: What is SPI Flash ID: 10 dc 1    ?

oe3jtb, 2016-04-29:
when trying to write users.csv to a MD380 UHF I get

SPI Flash ID: 10 dc 1
local variable 'size' referenced before assignment

I think this part is responsiible

def flashgetid(dfu):
    buf=dfu.spiflashgetid();
    print "SPI Flash ID: %x %x %x"%(buf[0],buf[1],buf[2])
    if (buf[0] == 0xef and buf[1] == 0x40):
        if (buf[2] == 0x18):
            sys.stdout.write("W25Q128FV 16MByte\n");
            size=16*1024*1024;
        elif (buf[2] == 0x14):
            sys.stdout.write("W25Q80BL 1MByte\n");
            size=1*1024*1024;
        else:
            sys.stdout.write("Unkown SPI Flash - please report\n");
            size=0;
    return size;

The procedure runs direct to "return size" without setting "size" to a value

Any suggestions?
tnx kind regards
Alex

travisgoodspeed, 2016-04-29:
Howdy Alex!

The UHF model of the MD380 usually has a 16MByte W25Q128FV chip, but the VHF model carries a smaller 1MByte chip.  Your radio has a model that we haven't seen before.  Congratulations!

Could you send us some photos of your radio, or tell us more about its branding (Tytera/Retevis/TYT/NKTech/etc)?

We'll add your chip to the table in short order.

--Travis

oe3jtb, 2016-04-29:
Hi Travis
tnx for the fast answer
so here it is
http://i.imgur.com/iziTKSt.jpg
http://i.imgur.com/Sqv3V9l.jpg
tnx
kind regards
Alex

dg3yjb, 2016-04-29:
Hello,
the same problem is here.
Branding is Tytera.

[image]

rct, 2016-04-29:
Just curious does that model ID make it into the .rdt or bare code plug?

dg3yjb, 2016-04-30:
A Screenshot from the original tyter MD-380 programming tool
[image]

aeickho, 2016-05-01:
corr python error with: c52cd0641b5615152ef85056c423f10bc2517f9e

aeickho, 2016-05-01:
one solution see: @travisgoodspeed  #66 try and error to detect the spi-flash size.

oe3jtb, 2016-05-01:
[image]

Took a photo with my electronic microscope, should be the 16MB, but with other ID

aeickho, 2016-05-02:
On e06be0fb1385d09f7dbd003f825cded5b728e072
is a test version with interpret  Flash ID: 10 dc 1 as 16MByte spi flash chip

oe3jtb, 2016-05-02:
Hi
did a new clone
git clone https://github.com/aeickho/md380tools.git

and than the same as before
now the output from 
./md380-tool spiflashwrite data 0x100000
is

SPI Flash ID: 10 dc 1
can't programm spi flash wrong flash type

aeickho, 2016-05-02:
git clone https://github.com/aeickho/md380tools.git
git checkout next_spi_flash

make clean

and than the same as before
now the output from
./md380-tool spiflashwrite data 0x100000
is

oe3jtb, 2016-05-02:
ok tnx

so in line 342 there is in my opinion a typing error
def flashgetid(dfu):
    size=0;
    buf=dfu.spiflashgetid();
    print "SPI Flash ID: %x %x %x"%(buf[0],buf[1],buf[2])
    if (buf[0] == 0xef and buf[1] == 0x40):
        if (buf[2] == 0x18):
            sys.stdout.write("W25Q128FV 16MByte\n");
            size=16*1024*1024;
        elif (buf[2] == 0x14):
            sys.stdout.write("W25Q80BL 1MByte\n");
            size=1*1024*1024;
    elif (buf[0] == 0x10 and buf[1] == 0xdc):
        if (buf[2] == 0x01):
            sys.stdout.write("W25Q128FV 16MByte maybe\n");
            size=16*1024*1024;          
    else:
            sys.stdout.write("Unkown SPI Flash - please report\n");
    return size;

in original 
elfi (buf[0] == 0x10 and buf[1] == 0xdc):
^^^
I corrected to elif

and run again
after serveral erase
the last lines of the output are

1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 125 11f400 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 126 11f800 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 127 11fc00 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 128 120000 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 129 120400 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 130 120800 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 131 120c00 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 132 121000 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 133 121400 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 134 121800 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 135 121c00 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2351 136 122000 1024 
1024
1033
[Errno 110] Operation timed out

tnx Alex

aeickho, 2016-05-02:
if you repeat flashdb ... stopped it on the same address?

aeickho, 2016-05-02:
please fetch the branch ... some additional function added: flash exclusive  now

oe3jtb, 2016-05-03:
Bingo !!! all is working fine, uploaded the whole world!!!

TNX
Alex, OE3JTB

aeickho, 2016-05-04:
I'll Patch ... The spiflash type test .. Then I close this issue..

closed 2016-05-04 by aeickho