travisgoodspeed/md380tools #80: Menu Label Cosmetic

aeickho, 2016-05-02:
At the moment, this texts are included:
static wchar_t wt_addl_func[]         = L"Addl. Funct";
static wchar_t wt_datef[]             = L"Date format";
static wchar_t wt_debug[]             = L"Debug";
static wchar_t wt_disable[]           = L"Disable";
static wchar_t wt_enable[]            = L"Enable";
static wchar_t wt_rbeep[]             = L"M. RogerBeep";
static wchar_t wt_userscsv[]          = L"UsersCSV";
static wchar_t wt_datef_original[]    = L"Original";
static wchar_t wt_datef_germany[]     = L"German";
static wchar_t wt_promtg[]            = L"Promiscuous";
static wchar_t wt_edit[]              = L"Edit";
static wchar_t wt_edit_dmr_id[]       = L"Edit DMR-ID";
static wchar_t wt_no_w25q128[]        = L"No W25Q128";

Please support me to find the correct words.

oe3jtb, 2016-05-03:
static wchar_t wt_rbeep[] = L"M. RogerBeep";
static wchar_t wt_datef_original[] = L"Original";
static wchar_t wt_datef_germany[] = L"German";

My suggestion
static wchar_t wt_rbeep[] = L"Smooth Beep";
static wchar_t wt_datef_original[] = L"YYYYDDMM";
static wchar_t wt_datef_germany[] = L"YYYYMMDD";

What does ?
static wchar_t wt_debug[] = L"Debug"
static wchar_t wt_promtg[] = L"Promiscuous";

73 Alex

dg9vh, 2016-05-04:
when german it is: 
static wchar_t wt_datef_germany[] = L"DD.MM.YYYY";

closed 2016-05-08 by aeickho

jruys, 2016-05-10:
I have

Date Format
YYYY/MM/DD
DD/MM/YYYY (I changed the code locally to use "/" in stead of ".")

Don't think there's an ISO name for the various modes. I'm sure the US folk will ask for MM/DD/YYYY