travisgoodspeed/md380tools #81: Flashing callbook to SPI flash on MD390

zarya, 2016-05-02:
After the libusb error radio reboots. Firmware runs without issues

2302 2301 33f400 1024 
1024
1033
(OK, 0, dfuDNLOAD_IDLE, 0)
2302  33f800 672 
672
681
(OK, 0, dfuDNLOAD_IDLE, 0)
Exception usb.core.USBError: USBError(19, u'No such device (it may have been disconnected)') in <bound method Device.__del__ of <DEVICE ID 0483:df11 on Bus 002 Address 006>> ignored

zarya, 2016-05-02:
zarya@mexicano:~/github/md380tools$ ./md380-tool spiflashwrite data 0x100000
SPI Flash ID: ef 40 18
W25Q128FV 16MByte
erase 100000

zarya, 2016-05-02:
Same error also on MD380

aeickho, 2016-05-02:
Witch branch you are using?

aeickho, 2016-05-02:
The reboot after flashing is new ..

zarya, 2016-05-03:
Branch master last commit:
zarya@mexicano:~/github/md380tools$ git log
commit b5d33817c4e0711133715473da991a90ab89d83b
Author: Alex Eickhoff <alex@artbanause.de>
Date:   Mon May 2 15:08:51 2016 +0200

    remove one warning

zarya, 2016-05-03:
All good now since the last commit (8ed433e):

2302  33f800 672 
672
681
(OK, 0, dfuDNLOAD_IDLE, 0)
reboot radio now

closed 2016-05-03 by zarya

aeickho, 2016-05-03:
@zarya tnx
...
(OK, 0, dfuDNLOAD_IDLE, 0)
2302 33f800 672
672
681
(OK, 0, dfuDNLOAD_IDLE, 0)

i think this is correct ... the db is flashed completed... there is now a reboot after flash 
 ccb1ce4fe3e7a7063e2b617fa0f1ea6f142d747c

aeickho, 2016-05-03:
fixed with 8ed433e702df05f5b1dd4301e05b4f28b906bb7f

paulomartinspt, 2016-07-30:
Hi @aeickho 
I tested MD380Tools today for the first time on Linux, i'm been using it but only flashed the firmware through the Windows normal process.
Everything went well with the FW, loaded todays build but when running the users.csv command to upload it to the radio, i receive the same error that @zarya received

wc -c < data.csv > data
cat data.csv >> data
./md380-tool spiflashwrite data 0x100000
SPI Flash ID: ef 40 18
W25Q128FV 16MByte
erase 1981729 bytes @ 0x100000
flashing 1981729 bytes @ 0x100000
reboot radio now
Exception usb.core.USBError: USBError(19, u'No such device (it may have been disconnected)') in <bound method Device.__del__ of <DEVICE ID xxxx:xxxx on Bus 003 Address 014>> ignored

Does this mean it went well or not?
I assume that if he reboots, it's updates correct?

Regards,
Paulo

Latest Ubuntu build and updated
Latest PyUSB
Todays build of MD380Tools
MD380 Old Encoder