travisgoodspeed/md380tools #82: Introductory Video by LadyAda

rct, 2016-05-03:
Limor Fried, recently licensed as AC2SN, has an intro video on the MD-380 and installing the experimental firmware on Windows with Cygwin.  Unfortunately, it's really basic as she/(they) are filming and discovering at the same time, but it might be useful to some. Close this issue when it is determined if it worth linking to this in the README.md or not.

https://www.youtube.com/watch?v=mAS3nH6wuNk

maushammer, 2016-05-04:
I noticed the new fonts didn't take for her -- md380-gfx is run correctly from the makefile, but it doesn't find any characters to replace. The python crc32 is used for a checksum, so this should have worked...

ladyada, 2016-05-04:
YAY CYGWIN! 
I could try MSYS but man i'd prefer not to
i think the checksum fails for some non-python reason but since i wanted to continue with the video i just commented out the checksum and it managed to upload. can give you more concrete error evaluation if ya want it

dl8bb, 2016-05-04:
As shown in the Video the problem is the line ending with '\r' but cygwin like to deal with '\r\n' instead.

The checksum is normally not necessary but from time to time we saw broken downloads from the blog site where the firmware zip file is hosted.

ladyada, 2016-05-04:
i *think* its a remnant of git clone under cygwin and CRLF conversion. mostly because its a bane of my life under windows

ajorg, 2016-05-04:
Is the checksum operating in text mode instead of binary mode then?

ladyada, 2016-05-07:
yah fixed it by adding dos2unix in the Makefile (basically, md5sum -c chokes on the \r in the md5 file)
https://github.com/travisgoodspeed/md380tools/issues/85

admins, you can close this issue so its not duplicate

travisgoodspeed, 2016-05-09:
Closing this issue, with further Cygwin compatibility discussion in #85.

closed 2016-05-09 by travisgoodspeed