travisgoodspeed/md380tools #85: md5sum checksum compare don't work on cygwin

aeickho, 2016-05-04:
see #82.. 

@ladyada reported the md5sum checksum compare from @dl8bb don't work on cygwin

6945c79f455941f16e7b996304d759c7407d0976

aeickho, 2016-05-04:
Without an Cycwin environment  I can't fix it :(

ladyada, 2016-05-07:
hiya so i have a fix for this, basically change firmwares/makefile to add "dos2unix" for the md5 files since otherwise md5sum can't parse them. can do a pull req or you can just add the lines, or if you have another suggestion lemme know :)

`D002.032.bin: D002.032.zip
	@unzip -p D002.032.zip 'Firmware 2.32/MD-380-D2.32(AD).bin' >D002.032.bin
	dos2unix D002.032.bin.md5
	@if md5sum -c D002.032.bin.md5 2>&1 > /dev/null; \
	then echo "MD5 OK"; \
	else \
		echo "Firmware 2.32/MD-380-D2.32(AD).bin wrong md5sum"; \
		rm D002.032.bin; \
	fi;

D002.032.zip:
	curl --help >>/dev/null #Check that we have CURL.
	curl 'http://www.va3xpr.net/?ddownload=9197' >D002.032.zip
	dos2unix D002.032.zip.md5
	@if md5sum -c D002.032.zip.md5 2>&1 > /dev/null; \
	then echo "Download OK"; \
	else \
		echo "Download wrong md5sum"; \
		rm D002.032.zip; \
		exit; \
	fi;`

travisgoodspeed, 2016-05-09:
@ladyada Please send a pull request with the tested changes, and we'll merge to keep Cygwin compatibility.

Is there anything we can do to make the Python tools for USB work on Windows?  I don't think libusb works in Cygwin or Windows Services for Linux.

ladyada, 2016-05-09:
if you mean the user.csv-uploading-tool, yes i did get it working! you just have to make a libusb 'holder' driver and zadig or manually switch drivers - i havent finished documenting yet but it did work. 
if you mean something else, lemme know & i'll hack on it

73 AC2SN (<- am cool ham radio person now)

travisgoodspeed, 2016-05-22:
I had to revert #132 because it broke compatibility in Unix.  Rewriting the Makefile now to clean things up; the rewrite ought to run on Cygwin.

travisgoodspeed, 2016-05-22:
Pushed a new Makefile that checks the file sizes without using md5sum.

@ladyada, would you please verify that this works on Cygwin?  If so, I'll close the issue.

ladyada, 2016-05-22:
tested, works fine on cygwin

travisgoodspeed, 2016-05-22:
Excellent!  Closing this issue.

M0IVQ, 2016-07-30:
Commit dad4210 breaks this fix on Cygwin due to shasum -a 256 being sha256sum on Cygwin

travisgoodspeed, 2016-07-30:
Ugh.  Close this issue when Cygwin works again.

Is there a continuous integration service to test for compatibility with Cygwin and GNUWin32?  Without such a thing, we'll keep breaking Windows compatibility as we develop on Linux and Mac.

d235j, 2016-07-30:
The shasum command comes with perl; do you have it installed? That said, autodetection and use of whichever sha2 command is available would be ideal.
I intend to do more tests with MSYS2; as for CI there's AppVeyor and others; I can set something up.

travisgoodspeed, 2016-07-30:
Cool.  Assigned to @d235j.

Please close this issue when (1) the immediate problem is fixed and (2) we have some sort of CI to prevent future compatibility breaks with Cygwin.

M0IVQ, 2016-07-30:
I'll try installing the Perl-Digest-SHA package tomorrow and give it another try. If it works, I'll create some instructions.

M0IVQ, 2016-07-31:
With the Perl-Digest-SHA package installed, firmware now builds without errors.

davehun, 2016-08-03:
would a docker image for the toolkit be a sensible idea ?

rct, 2016-08-03:
A Dockerfile that builds an image with the necessary tools to build the firmware and then builds the firmware would be a useful contribution to add to this repo.  However, it won't necessarily solve all the problems for those who are on Windows.   

To flash the users.db they either need to have the python bits and the USB libraries on windows, OR the docker VM needs to be given access to the USB port, so there needs to be coordination with the virtualization platform on windows (VirtualBox for the older/standard docker, and hyper-v for windows 10 with the new docker-on-windows bits).

So as far as this specific issue goes. I haven't really looking into it yet, but given that there is already a Python dependency md380tools, I think doing the hash check in Python using hashlib would cause the least amount of issues.   (the less dependencies on *nix/posix environment under Windows the easier it would be.

https://docs.python.org/2/library/hashlib.html

So I think it's basically
```
import hashlib

b = file("D002.032.bin").read()
sum = hashlib.sha256(b).hexdigest()
```

M0IVQ, 2016-08-03:
As there is already a need to install the git and curl packages as part of the Cygwin install, there isn't really issue with also installing perl-digest-SHA for the shasum commands to work. Like the GNUwin32 instructions, I think we just need to include instructions for using Cygwin including specific dependencies. It was my lack of knowledge with Cygwin that lead to this issue being reopened. I am in the process of writing a quick set of instructions but I need to test them on a fresh install before committing.

rct, 2016-08-04:
Ok, but It seems unnecessary to introduce a dependency on Perl and a Perl module for checking a hash, especially when there is already a hard dependency on Python for md380tool.  (If there are other parts of the md380tools package that use Perl, the point is moot.)

M0IVQ, 2016-08-31:
@travisgoodspeed wrote:
> Is there a continuous integration service to test for compatibility with Cygwin and GNUWin32? Without such a thing, we'll keep breaking Windows compatibility as we develop on Linux and Mac.

I'm currently working on a config file for AppVeyor CI to check GnuWin32. Once I have it working I'll submit a pull request and see if there is anything I can do for Cygwin.

travisgoodspeed, 2016-08-31:
Is the original issue still a problem, or can we close this issue?

@m0ivq Excellent to hear that you're working on an AppVeyor definition. It'll save us a ton of support requests.

M0IVQ, 2016-08-31:
The problem is solved by installing Perl-Digest-SHA but @rct raised a point whether or not perl is only used for checking the hash. If perl was chosen for a reason or is used elsewhere then it's safe to close this issue.

m0ivq, 2016-09-05:
@travisgoodspeed I think we can close this issue and open a separate issue for continuous integration support for Cygwin. AppVeyor.yml file supporting GnuWin32 has been merged so is ready if you wish to add an appveyor account. Config has been tested in my repository.

travisgoodspeed, 2016-09-05:
Closing this issue.  #259 will track the Cygwin CI stuff, and I'll figure out an infernal appveyor account.

closed 2016-09-05 by travisgoodspeed