travisgoodspeed/md380tools #89: 1750 Hz tone

der-stefan, 2016-05-04:
Add the option to send a 1750 Hz tone by pressing one of the side buttons.

aeickho, 2016-05-04:
I think, this is possible - not now but in the near future

der-stefan, 2016-05-05:
The DTMF tones are generated in the STM32 and feed out on PC8 (pin 65) and feed into the C5000 on MIC2_P (pin 5). The same path could probably be used for a 1750 Hz tone.

travisgoodspeed, 2016-05-05:
Thanks, Stefan!  If PC8 is raw audio, we might be able to use it for APRS transmissions on the VHF model!

(But dear Lord, don't create an issue for that until we have this much working.)

der-stefan, 2016-05-05:
@travisgoodspeed I haven't measured the pins, my statement is based on the schematic. And it would make sense: The STM32F405 has a timer (TIM8, CH3) there. Directly after the pin comes a small lowpass filter. I bet, the pin is already used in the function of a timer for creating the DTMF tones.

der-stefan, 2016-05-08:
There could also be the possibility to completely bypass the C5000 by using the CTCSS/DCS output of the STM (PC7, Pin 64). This signal is directly feeded to the modulator! So there is the chance of generating signals only with the microcontroller.

aeickho, 2016-08-27:
the 1750 Hz tone is added by Retevis with d13.020 ... enable it on cps 1.32

aeickho, 2016-08-27:
see #218

closed 2016-08-27 by aeickho

der-stefan, 2016-08-27:
Great news! Perhaps the Chinese read our issue here...