travisgoodspeed/md380tools #90: Status LED signaling

der-stefan, 2016-05-04:
It would be a nice feature to blink the green LED every 10 seconds to show that the radio is switched on. Since the display turns off after a certain time there is the risk of forgetting the radio, until it turns itself off by a drained battery.

der-stefan, 2016-05-13:
Now it happened to myself: Over night my "Tüte" was switched on accidentally, because I didn't realize it. Either the LED should blink every few seconds, or we should have the option to dim the screen.
(Hopefully my battery was not deep discharged!)

travisgoodspeed, 2016-09-23:
Merged into #370.  Reopen only if there is active progress in adding such a feature.

closed 2016-09-23 by travisgoodspeed

Wolf-DL4YHF, 2017-02-20:
Trivial to implement now since there is a 'hooked' SysTick_Handler() for the backlight control, and (coming soon) Morse code output. Is there still some interest in this feature ?
( I personally prefer the dimmed backlight, to see the radio is still on )