travisgoodspeed/md380tools #93: Show RSSI value instead of receive icon

der-stefan, 2016-05-05:
The icon in the upper left corner is somehow useless. According to the manual it should indicate the reception. It would be a nice feature to write the RSSI value instead. In the future we could measure the RSSI value and calculate a power level in dBm from it.

der-stefan, 2016-05-08:
There is a document by KA6BFB about the RSSI output of the TA31136:
http://www.satellitearc.com/images/zeroizerTA31136.pdf

aeickho, 2016-05-08:
very interesting, but I have not found how and where the converter is read

aeickho, 2016-05-08:
0x0802079c is the place to create the ant symbol

ladyada, 2016-05-25:
i was curious about the rssi too, so i popped open my MD380 and probed the ST's pin #35  (RSSI input) to see what the signal looks like. 

[image]

When you first boot and/or go into and out of the radio, the signal is a nice clean DC which is also probably the true RSSI voltage detect:

[image]

After a few seconds, this signal turns into a ~10Hz squarish wave, where the peak voltage is the same as the original DC RSSI signal, but ~ 40% duty cycle. I haven't desoldered my radio mainboards so I haven't looked at the opposite side of the 1K series resistor - so unclear if the ST is somehow driving the pin low, or if the signal really is that whack right out of the detector. Adding a 0.1uF cap didnt help.

[image]

could be best option is to fill a small buffer of analog readings (in the rtc hook?) and pick the peak value? who knows, just messing around :)

maushammer, 2016-05-25:
Nice!  I take it that's receiving an active signal that's being played out the speaker, right?

The TDMA time slots are 30ms each, with a sync in the middle.  Those peaks look like 15msec, so that might make sense, but the 10 Hz doesn't ... unless it's not receiving and it's some sort of power-save mode.
[image]

http://www.etsi.org/deliver/etsi_ts/102300_102399/10236101/02.02.01_60/ts_10236101v020201p.pdf

der-stefan, 2016-05-25:
By the way: The C5000 itself can also be configured to analog FM (see https://github.com/pchickey/md380-re/blob/master/datasheets/hrc5000.pdf). And it has a register (or two for high and low 8 bits...) to read out a RSSI value. Without measuring we can't say if the TA31136 or the C5000 is used for FM.
Questions:
- Why did they implement two ways for analog FM?
- Could we use both transceivers at the same time? At least for receiving?
- What does the "funny" receiption icon in DMR mode mean? How is it derived: Read out of some C5000 registers or read out of the TA31136?

ladyada, 2016-05-25:
k, those measurements above are not during any audio receive, just idling on a local DMR repeater (but no audio transferred). so perhaps it is a power saving thing. 

i made a typo above, if you enter/leave the *menu* system the square-wave resets to DC. this is why when you go to the menu and back to the main screen you suddenly get the right RSSI signal again. the peak amplitude of the square wave changes based on what antenna i use.

travisgoodspeed, 2016-05-25:
The RSSI register in the C5000 is only valid for digital packets, and only during that packet's arrival.  (That is, you must fetch it before clearing the buffers.)  It does not work for analog transmissions.

@ladyada What do you see on the scope when receiving an analog FM signal?  Does it also correlate with what you see as the RSSI on the screen?

I'll patch the incoming packet handlers to print the RSSI value into dmesg.  If it looks reliable, we might consider using that as a source to replace or augment Pin 35's signal for digital packets.  For analog, where the C5000's RSSI register isn't updated, I think we're stuck with the TA31136's output.

ladyada, 2016-05-25:
OK, w/radio set to analog channel...

  - When monitoring (M button pressed once): DC signal which will vary if i switch or remove the antenna. Stays consistant, no peaks, during entire monitoring period

  - When idle, not monitoring, but channel select on analog: for first 5 seconds, DC and sometimes with little peaks (these peaks also appear when on a digital channel, dont appear during active monitoring)

[image]

after 5-6 seconds, goes to a squareish wave

[image]

For some reason last night the square waves didn't have the peak variation. not sure what changed this morning, battery is still topped up

@travisgoodspeed oh yah forgot to mention, the onscreen antenna bars drop out at the exact same time the signal goes from DC to square - both on digital and analog. if im on an analog channel, and monitoring, the rssi signal is a nice flat DC forever and the antenna bars stay stable forever. so i dont see any evidence that the onscreen bars use anything other than the RSSI analog signal, they're basically identical.

travisgoodspeed, 2016-09-23:
Merged into #370.  Reopen only if there is active progress in adding such a feature.

closed 2016-09-23 by travisgoodspeed

sijskes, 2016-10-06:
netmon displays the antenna icon value in 'sm:'. observations: when using stock antenna, inside the house. values random between 3...5. when 'nosig'. connected to outside antenna, values near 0 on 'nosig'.

brafi, 2017-05-08:
Any way to see the RSSI value on the latest firmware?