travisgoodspeed/md380tools #96: Radio hangs on system start

aeickho, 2016-05-06:
sometimes, after power on, the screen is white, no rx no tx is possible

travisgoodspeed, 2016-05-06:
Have you found a way to reliably trigger this bug?

aeickho, 2016-05-06:
I have the first time seen the error before about 4 weeks

aeickho, 2016-05-06:
"1/20 times the error occurs" not now :(
30 restarts of 2 radios now "white-screen"

travisgoodspeed, 2016-05-06:
Perhaps a recent patch is causing trouble, or stalling a thread?

aeickho, 2016-05-06:
now I reboot the radio via usb ...

aeickho, 2016-05-06:
Idea: I's  occurs only after firmware update

aeickho, 2016-05-06:
@all can you report an "white screen" on an radio with stock firmware ?

maushammer, 2016-05-06:
I've seen it twice out of maybe 70 power-ups with minimally-modded firmware (font changed only - I haven't applied any code patches)

Wild guess: does radio reception at time of powerup have anything to do with it?  Traffic is pretty light on my local repeater.

M0IVQ, 2016-05-06:
I have seen this on stock firmware - if radio receives just after power on then locks up with white screen.

aeickho, 2016-05-08:
If there are no more comments, I'll close this Issue in 3 days

phl0, 2016-05-08:
I cannot confirm this issue when radio receives immediately after powering the radio. Switched it off and back on a few times while transmitting with a second radio and all working fine.

oe3jtb, 2016-05-09:
Have it also somtimes, also with stock firmware, my explination was quick switchoff and switch  on

travisgoodspeed, 2016-05-11:
I've seen it once or twice, but I haven't been able to reproduce it.  If we don't come up with a reproducible way by tomorrow, let's call it an act of god and close the issue.

commit 75bc2647 (2016-12-09): add the "new" display init also to the non gps radios #626

fixed #96 maybe, new display_reset
commit adb9c544 (2016-12-09): add the "new" display init also to the non gps radios #626

fixed #96 maybe, new display_reset
closed 2016-05-16 by aeickho