travisgoodspeed/md380tools #98: Volume control circuitry

tristantech, 2016-05-07:
The MD380 schematic shows that the volume knob / power switch is directly adjusting the VOL_OUT signal before it goes to the AF amp as opposed to digital volume control. If this is the case, does anybody know how the CPU is able to read the volume level and display it on-screen?

aeickho, 2016-05-08:
this is a open item for a long time from me. 
the "public" schematic is  not true :( see red LED 

see @ 0x0800df84 there is the "volume" menu

on ram unit8_t *  0x2001d3b5 is the actual position of the volume knob

but I can't locate the position that’s write to this   address

tristantech, 2016-05-09:
Do you think there's a chance that the radio does its AF processing digitally? (there is an I2S out from the HR_C5000 and audio-capable DACs on the micro) It does seem a little unlikely given how remarkably similar much of this radio is to other products like the CS700 or some of the application notes out there ([1](http://www.52rd.com/Blog/Detail_RD.Blog_chinarf_78528.html), [2](http://www.ha-jx.com/mod_article-article_content-article_id-204.html)).

I briefly looked over the board to spot for digital pots or similar but saw nothing obvious (although I can't ID most of the smaller packages)

There are many other things on the "schematic" that make me very skeptical, like the DMR and PLL SPI ports that connect around PE2-5 (those pins are not even a SPI port on this micro). Do you know where this schematic came from?

aeickho, 2016-05-21:
See also #116

eb4fbz, 2016-07-04:
Volume control is 100% analog and pot position readout is very noisy and prone to errors. I suggest to remove the volume bar from appearing on the screen and perform attached hardware mod so beep tones are injected at the volume pot input. Mine is working with this modification since 11/2015.

Remove R341 (100K) and connect the signal to VOL_OUT (R253) using a 47K resistor and wrapping cable.
[image]
[image]

travisgoodspeed, 2016-07-04:
Well, that settles that.

We'll close this issue when @eb4fbz's hardware mod is properly documented somewhere that won't bitrot.  (Github issues bitrot.)

Cheers from Philly,
--Travis

Edge123, 2016-07-13:
So, stupid question.  I did this mod successfully on 2 radios (worked great, thanks @eb4fbz), last one i launched the 47k resistor into space, and its gone.  Is it needed (i am assuming it is), and what size is are these so i can order a few replacements.

eb4fbz, 2016-07-13:
Resistor size is 0402.

eb4fbz, 2016-09-04:
Volume pot position is read through pin 23 (ADC123_IN0). It should read a DC voltage between 0.1 and 1.6V, but there is also AC noise coming from HR_C5000 audio output that is not fully supresed by the RC low pass filter.

A hardware mod to disable these readings would imply to move a (0402) 22K resistor to ground, so ADC allways read 0V.

I have extracted the schematic and have it written down on paper, give me some time to clean it up.

Regards,
Carlos EB4FBZ

closed 2016-12-23 by travisgoodspeed

donde1, 2017-03-19:
I have the same problem with the newer **MD-390**.  It's about a week old. Thought I would make a small 2.5 mm plug and jack attenuator box. Man, this is such an obvious problem. TYT should fix it with a firmware update!