travisgoodspeed/md380tools #100: please can you add this as a bootlogo

basmes, 2016-05-08:
Hello travis 

Please can you make this file as a bootlogo file for the md380. 
I don't now how to make them by myself. i only now how the change the logo's that are located in the patch folder. i don't have the experience. I tried  whit GIMP en chance te makefile but it fails.
please can you make it.

thanks 

greetz bas

[image]

[image]

travisgoodspeed, 2016-05-08:
I'm sorry, but I'm too busy to do logo requests. You might try asking for help in the #md380 channel on Freenode IRC.

Closing issue.

closed 2016-05-08 by travisgoodspeed

basmes, 2016-05-08:
thanks 

for the information

greetz

bas

mach327, 2016-05-09:
@basmes I have a small script to handle the comments and a little bit to add to the README that should aid others in customizing their boot logo. Will try to push it soon, maybe later today.

basmes, 2016-05-09:
thank you 
i see it in readme.md at the end

i will try it today

./gfx-header copy patches/2.032/0x80f9ca8-poc.ppm ./your_image_file.ppm

is this a command in linux terminal  of a line inte makefile ad the patched folder.

greetz 

bas

basmes, 2016-05-09:
thank you 
i see it in readme.md at the end

i will try it today

./gfx-header copy patches/2.032/0x80f9ca8-poc.ppm ./your_image_file.ppm

is this a command in linux terminal  of a line inte makefile ad the patched folder.
i have changend the bootlogo whit 0x80f9ca8-shmoo
an that works in the makefile.
if i type youre rule in the terminal it says file are map doesn't exist

greetz 

bas

mach327, 2016-05-09:
That's because I haven't pushed it back to this repository, so your copy doesn't have that file.

You can fix the comments issue manually by opening the PPM images in a text editor, and copying the comments from a good image to your image.

basmes, 2016-05-09:
[image]

thanks it works