travisgoodspeed/md380tools #103: arm-none-eabi-gcc command not found

webkernel, 2016-05-09:
Hi,

I am using Debian 8.2 out-of-the-box install and after make clean diet i got this error:

arm-none-eabi-gcc command not found

Please help.

dl8bb, 2016-05-09:
You need to install the appropriate arm compiler suite. Please take a look i.e. https://launchpad.net/gcc-arm-embedded or install the debian packages for arm build environment. 

_sudo apt-get install gcc-arm-none-eabi_

You need to have a working build environment with make, linker etc already installed.

If you don't have it, install at least the build-essential package.

_sudo apt-get install build-essential_

Just my 2 cents

travisgoodspeed, 2016-05-09:
Close this issue when README.MD contains explicit apt-get instructions for Debian.

webkernel, 2016-05-09:
many thanks. it worked!

closed 2016-05-09 by travisgoodspeed

LW-Ho, 2018-08-15:
Thanks a lot. it worked!

1399999, 2022-03-29:
Big thanks. It worked.

commit 73b3487e (2016-05-09): Merge pull request #106 from hax404/master

README.md: add required packages for debian.  Close #103 .