travisgoodspeed/md380tools #104: Wiki or User Group?

gadget-fan, 2016-05-09:
OK I have been following this project on github for a while now and I am thinking I am ready to try it out, but right now I feel like I could use more hand-holding that what is in the readme files.  Has anyone started compiling a Wiki or done some write-ups in a user group somewhere?  First order of business of course is that I need to compile everything and load firmware and I can get some of that from the readme docs, but then I don't see much documentation of changes and features, at least not in one place and easy to read.  Any help out there?  Could we get something started?  I would be willing to help but warning I couldn't put in a ton of time right now because of other commitments (like the rest of you have nothing else to do right).  I know documentation is the worst so I feel your pain!  If you can point me in the right direction or tell me how I could help get something started that would be excellent!

travisgoodspeed, 2016-05-09:
Github offers a wiki or website, but I'd prefer the user's group to be managed separately from the development.  It gets a bit noisy with support requests showing up in the Issues tracker.  Could someone volunteer to start a user's group in another venue?  (Facebook group, mailing list, wiki.)

Close this issue when we have a place to politely send those who need help.

gadget-fan, 2016-05-09:
What about using the github wiki but all Q/A goes to a yahoo/FB/google group?  I did some quick looking today and it seems easy enough to set up a group in any of those three but I don't see that any of them have wiki capabilities for documentation.  I totally get your point and agree, in fact I saw you field a request that would have been better suited for a user group today.

ae5au, 2016-05-09:
I like what @gadget-fan is thinking.  A wiki on GitHub as the authoritative documentation is better than trying to sift through any of the other options to get up to speed from scratch.  My vote for the public forum would be a Facebook group.  I get more than enough email already. Yahoo and Google groups are a pain to manually check without setting up email notifications.  This is all just my preference, of course.  I'll go along with whatever. :)

mach327, 2016-05-09:
Gitter is meant more for developers it seems, but the format seems suitable for Q&A/live help/user group. Maybe worth a look.

dg9vh, 2016-05-10:
Facebook needs also an account and want's realnames etc... so I think it is not really an option for everybody.

Why not starting a google group? It's easy and comfortable to use per e-mail, also from mobile phones.

just my 2 cent

YV5OF, 2016-05-11:
Google Group --> my vote +1

travisgoodspeed, 2016-05-11:
Cool.  Please start a user's group and we'll add a link to it in the README.

dg9vh, 2016-05-11:
Just created a google group https://groups.google.com/forum/#!forum/md380tools thats public open at the moment. So no registration should be necessary.

travisgoodspeed, 2016-05-11:
Closing this issue now that the group is mentioned in the README. 

Thank you kindly for taking care of it.

closed 2016-05-11 by travisgoodspeed