travisgoodspeed/md380tools #105: Menu/Text Timeout

travisgoodspeed, 2016-05-09:
The menus and text entry have a rather short timeout, which can be infuriating when writing a text message.

Close this issue when we have a menu entry to optionally disable this timeout, so that menus need to be manually exited and text entry isn't cut off by a moment's distraction.

aeickho, 2016-05-09:
the timeout on my radio is 20sec. Is your timeout shorter?

travisgoodspeed, 2016-05-09:
Mine is also twenty seconds, but it makes text messaging frustrating.  If we eliminate the timeout and add predictive text database, then messaging on the MD380 might be as painless as on a Motorola or an old cellphone.

The Motorola radios will eventually dim their screens, but a text message in progress is not abandoned on timeout.  (Similarly, it doesn't exit menus if you let it timeout.)

aeickho, 2016-05-09:
currently I have no idea how the timeout is being implemented now. 0x8044662() is a big mystery for me

travisgoodspeed, 2016-05-09:
I'll race you to it, then!

M0IVQ, 2016-05-09:
Theres already a menu timeout setting in the CPS.

M0IVQ, 2016-05-09:
I believe its called 'Menu Hang Time' though cant remember exact location of it as away from home at the moment.

aeickho, 2016-05-09:
[image]

aeickho, 2016-05-09:
this is an nice information ....   this info make it easy to find the  place in the radio

but I stop working for today

M0IVQ, 2016-05-09:
I think if set to zero, menu doesn't time out (I've set mine to not timeout but can't remember whether setting is zero or whether there's an option for none). Would definitely be useful being able to change from front panel though.

der-stefan, 2016-05-09:
@travisgoodspeed A T9 like function would indeed be a great idea! But the database will be quite huge, especially if you want to have different languages. I have no experience what size a word database normally has.

aeickho, 2016-05-09:
at present: the Chinese T9 is  included yet in the firmware ...  see edit messege ^ Key twice .... we can remove it for more "creative" space

der-stefan, 2016-05-09:
You are completely right! There is a Chinese text completion on it. Press the "#" key in SMS mode to try it out. Can you give us a number, how much memory this dictionary needs? Let's remove the Chinese one and add English and other languages!

dg9vh, 2016-05-10:
For info: In CPS the menu-hang-time set to "hang" results in no timeout for editing sms and no timeout for menu... so no need to implement something in this way...

travisgoodspeed, 2016-05-10:
Excellent!  My apologies for any wasted time. 

Closing issue.

closed 2016-05-10 by travisgoodspeed

der-stefan, 2016-05-10:
@dg9vh This is good news. But still it would be a good feature to do this setting via the menu. But ok, it's not necessary...

dg9vh, 2016-05-10:
@der-stefan @travisgoodspeed I think this is a "lower priority"-feature, because it's done by CPS itself. Ok, it would be a "nice to have" I think, but I think other features like https://github.com/travisgoodspeed/md380tools/pull/83 would be better to merge because this gives us possibility to import codeplugs from original CPS.