travisgoodspeed/md380tools #109: Private Call

YV5OF, 2016-05-11:
Hi Team,

I have two versions of the FW compiled, one 05/09 and two 05/10.

I'm trying to make a private call to another radio, dialing numbers manually or contact book and it is not generated or not set.

You can only make public calls to engage reflector o TG.

Testing with TYT MD-380.

thank.

travisgoodspeed, 2016-05-11:
Works fine for me with the current master version.  On the MD380, you make a private call by selecting the name in the address book and hitting the PTT button, or by typing the number and hitting the PTT button.

What exactly do you do to cause it to fail?  Were you able to make private calls on the official firmware?

YV5OF, 2016-05-11:
Hi,

1-. With the official firmware works.

2-. With firmware compiled:
By selecting a contact or dial numbers and pressing PTT, TX starts but the screen does not show that this made the private call, or is doing an outgoing call.

The display remains the number to be dialed or selected contact, no changes to the screen displaying the outgoing call.

Thank.

travisgoodspeed, 2016-05-11:
Okay, so the call is made but the menu isn't exited, so the screen doesn't indicate the call in progress.

This is a side effect of a feature that we added to prevent incoming calls from interfering with menus and text messaging.

YV5OF, 2016-05-11:
Ok, 

Many thank my friend for replay.

73!.

closed 2016-05-11 by YV5OF