travisgoodspeed/md380tools #114: Output of DMR contact data cut off

phl0, 2016-05-11:
On incoming call the green box shows only 5 lines whereas if the call is over the color changes to grey and 6 lines are being displayed. Have been playing around with gfx.c for 2 hours now but couldn't identify a bug. Any hints to where I have to look?
[image]
[image]

travisgoodspeed, 2016-05-11:
As best I understand our current progress, we haven't yet hooked the code that displays the incoming call routine.  The CSV display is shown over it, but do to thread race conditions, it is very often overlaid by the bottom and top artwork of the screen.

Close this issue when the hooking for the incoming (and outgoing) calls is understood well enough to prevent visual artifacts.  This may or may not involve replacing the screen with one of our own design.

phl0, 2016-05-12:
No problem at all. I just made my first steps and tried to code a little and could not find a hint on where the green area is painted. Maybe I have to dig a little deeper into the sources and understand the stuff.

travisgoodspeed, 2016-05-12:
To understand that part, you need to dig into where the blue is drawn in the binary, not the source. We have to reverse engineer these things before we can move them to C for easy maintenance. 

It's annoying, but it keeps things exciting!

phl0, 2016-05-12:
Sure. I do know that now that I had another look a the C sources. I will
continue to try to understand what you are doing and how to "inject"
code into the binary firmware. And maybe one day ... :-)

travisgoodspeed, 2016-07-08:
I think this was fixed during other display cleanups.  Closing issue.

closed 2016-07-08 by travisgoodspeed