travisgoodspeed/md380tools #116: The next steps for this project : some suggestions

dg9vh, 2016-05-14:
Relayed from google-group:

> Hi again from Adam Kb2Jpd in Brooklyn NY
> 
> Thank you for your prompt response to your group.
> 
> There is a opportunity for you to make legitimate your patches for the Tytera MD-380/390 firmware that you created. If you create much needed features, the original creators would then feel obligated to enhance the factory firmware with your patches and set them in stone by including them in their next update.
> 
> There is a need for a setting on the menu or on the two external function buttons when both are depressed awhile powered on.
> 
> The setting is for ham amateurs who are blind , who need auditory feedback on the menu and the top scroll button. At first, just simple tones, then in the next generation, have Morse code spell out the contents of the screen. This should include a periodic information message in Morse when the radio is idling or scanning of the battery charge level. 
> 
> 
> If this seems to be a lot of work, perhaps you could publish a memory map of the firmware where some of us could add needed features and add hacks without destabilizing the firmware.
> 
> Thank you again for your hard efforts in improving the firmware. The fonts and the added menu items help out immensely the usage and enjoyment of the MD-380 and sister platforms.
> 
> Adam Kb2Jpd

(Origin: [https://groups.google.com/d/msg/md380tools/akH28wR9Nuw/F25XmvNWFQAJ])

An information about this relaying would also be posted in the group.

Forts117, 2016-05-15:
Sadly I don't think the original authors would feel ANY obligation to include 3rd party fixes in their firmware. That's assuming there even are any more official updates for the 380...

dg9vh, 2016-05-15:
Well, think about the additions for whitestickers. This would be an improvement in my eyes.

kb2jpd, 2016-05-15:
I wouldn't know a white sticker is if it hit me in the face. What is a white sticker?

Adam Kb2jpd

kb2jpd, 2016-05-15:
Picasso said it first then Steve Jobs: Great artists steal.
If the code is done and it is sturdy, the original coders would not spend time rewriting the code. They would lift it and use it to save time and braincells.

kb2jpd, 2016-05-15:
One other comment noted: the firmware has a innocent bug with the volume control. It changes on random moments. It changes upon receiving a signal. Maybe we can hunt that down and kill it.

dg9vh, 2016-05-16:
@kb2jpd a whitesticker is a blind man, he has a usually a white stick.

The volume bug is tracked in another issue.

aeickho, 2016-05-16:
kb2jpd
> One other comment noted: the firmware has a innocent bug with the volume control. It changes on 
> random moments. It changes upon receiving a signal. Maybe we can hunt that down and kill it. 

This is very exciting .. that would be a signs it be that would make the volume change over software. Not like in the wiring diagram.

aeickho, 2016-05-16:
@dg9vh which issue do you mean

kb2jpd, 2016-05-16:
The ability to change the volume control by software. One idea would be to ramp the volume or create an automatic volume control (agc).

M0IVQ, 2016-05-16:
The volume control is done in hardware, with circuitry to the micro that 'reads' the volume setting from the potentiometer to display volume level on the screen - the bug where it appears that the volume changes on its own appears to be ADC noise from the pot being set between thresholds for two volume settings. Wouxun mobile radios have the same issue.

aeickho, 2016-05-16:
:) currently I have no idea how that works with the volume. the first step is to understand the way of working... 

@m0ivq you have double checkt this? you know how the potentiometer on the cpu depends? the  wiring diagram is not complied in this aria...

M0IVQ, 2016-05-16:
The following was posted in the Yahoo TYT group last September:

"Ok, i've just checked this, don't worry that much about this bug. The volume level is displayed for user information, but real volume control is analog. If the pot is turned up, you will hear the communications, despite the screen shows "Volume 0".

The CPU tries to measure the pot position by injecting a reference signal instead of audio while audio amplifier is muted. It does so using the CPU beep output, that is connected in parallel to the DMR demodulator AF output. This could be problematic, for example, noise from the demodulator output could false the reading."

aeickho, 2016-05-16:
@M0IVQ tnx

dg9vh, 2016-05-16:
@aeickho exactly I thought about https://github.com/travisgoodspeed/md380tools/issues/98 when writing that this issue is tracket.

My personal experience is that the volume level on the display is increasing but no audible increment of the volume in reality exists.

der-stefan, 2016-05-18:
Somewhere in the future we could start implementing a completely rewritten firmware. Right now we sew patches next to patches. One day the main part consists out of patches, but isn't as easy to maintain as a project completely written in C.

closed 2016-07-08 by travisgoodspeed