travisgoodspeed/md380tools #117: Problems with menu on VHF-edition

dg9vh, 2016-05-15:
Relayed from Youtube:

> Hi Kim and thanks for your video and help.
> I am new to the MD-380 (I have the Retevis RT3, VHF version) and I am having trouble with the new firmware. I installed the version of the firmware which you shared here (https://www.darc-saar.de/moziloCMS/index.php?cat=Relais&file=experiment.bin) but I can not change any settings in the menu (for instance I can not active german date option, or activate promiscuous mode). Any idea why? As I said I have a Retevis RT3 VHF radio. Tnx!

Anyone out there who can confirm?

aeickho, 2016-05-16:
my opinion: this  om is not working with the experimental fw. When flashing forget to activate the fw update mode

dg9vh, 2016-05-16:
@aeickho don't know - reasking him, what welcome logo he got. If he is using my edition (Saar-Edition) it would be a very impresive and absolutely clear logo that says "Saar Edition"... so I am waiting for reply from him.

kkomarica, 2016-05-16:
Hi ... I am the "youtube question guy" :) 
To answer the question ... 
Yes I get the logo and the extra menu. The radio info menu says: MD380 Tools ver. 2016-05-10

That said I tried flashing with other hacked firmware versions ... V1 and V2 from http://www.va3xpr.net/programming-software-firmware/ ... but they didn't flash correctly. I used the update program for windows supplied, and it said it was ok, but after flashing I did not get the new logo or the extra menu.
Maybe I should try different flashing (update) program ... if there is a different version?!﻿

dg9vh, 2016-05-16:
@kkomarica as mentioned on youtube: try http://www.va3xpr.net/?ddownload=9197 and extract the updater from that. To flash firmware, press at power on PTT and button above PTT to set TRX into flash mode.

kkomarica, 2016-05-16:
That's all clear, and I tried it several times with other versions. The only (partial) success was with the Saar version. All I can do is to try to flash it again. Is there a maybe a newer version of the saar edition?

dg9vh, 2016-05-16:
Yes, updated yesterday evening.

kkomarica, 2016-05-16:
Just updated to the newest firmware version. Flashing was successful but the original problem remains ... i can not change any settings in the menu ... for instance I can not active german date option, or activate promiscuous mode, or any change any other option in the MD380Tools menu

aeickho, 2016-05-16:
@kkomarica is it possible for you to make a video about the behavior.

dg9vh, 2016-05-16:
@kkomarica when flashing Codeplug into the transceiver, to **don't** have to put it into flash-mode, you simple plug in the programming cable, power on and start writing Codeplug into it.

kkomarica, 2016-05-16:
@dg9vh I've tried flashing again without entering into flash-mode. it works ok, but the problem remains.

@aeickho I will try to make a video tomorrow ... maybe I am doing something wrong :)

kkomarica, 2016-05-17:
here is the video of the strange behavior: 
https://youtu.be/WbadaG_xsM4

currently firmware version is 2016-05-13

aeickho, 2016-05-18:
see #69 
this is understandable behavior. I currently do not store any config in the vhf devices
i have no address on spi flash where i can be safely to write.

closed 2016-05-18 by aeickho

kkomarica, 2016-05-18:
ok tnx! 
just to be clear since i'm new to this radio and firmware ... the hacked/experimental firmware only works on UHF? VHF is not supported?

travisgoodspeed, 2016-05-18:
For safety reasons, many features are disabled on VHF for now.  We'll be bringing many of them back when Issue #66 is closed, pretty much everything but the DMR MARC user's database.