travisgoodspeed/md380tools #120: Admit criteria not respected

ae5au, 2016-05-16:
The factory firmware doesn't seem to respect admit criteria selections (at least for DMR channels).

Admit criteria of "Always" should always allow transmit.  This selection works properly.

"Color Code" should only allow transmit if there isn't another signal being received on the color code and time slot that the radio has programmed for that channel. In my testing (and that of others), it seems to behave just like "Always".  This is the largest problem as it allows you to transmit on a talk group that is free while there is traffic on another talk group that uses the same time slot.

"Channel Free" should not allow transmit if there is any other signal on the channel. I'm not sure if this should require the signal to be on the same time slot as programmed or not, but I would think that it should inhibit if any signal is being received...analog or digital.  In my limited testing, it does inhibit transmit if there is activity on analog or a different color code, but NOT if there is traffic on the same color code.

dg9vh, 2016-05-17:
I can confirm this issue and firmware behavior. @travisgoodspeed I think it should be tagged as enhancement.

M0IVQ, 2016-06-22:
2.034 and earlier stock firmware has this issue. Haven't yet tested on 3.008

travisgoodspeed, 2016-07-28:
Any word on whether this has been fixed in the 3.x series?

dg9vh, 2016-07-28:
@travisgoodspeed I just tried plain unpatched stock D003.008.bin - the issue is still existing, it is still not respecting admit-criteria color code... I can transmit over others...

xSmurf, 2016-10-18:
So I most definitely remember this working from my MD390G pre flashing. But I've now reverted back to both s13.020 and s13.012 and the politeness feature is still gone. I've now talked to at least four OMs who say they have noticed the same thing. I've replicated this in both simplex and repeater mode, and with the _Admit Criteria_ both as _Color Code_ and _Channel Free_. I've tried multiple code plugs as well, including a completely fresh one from the latest official MD390G CPS.

I'm at a loss, I swear it worked with the radio OOTB.

dg9vh, 2016-10-18:
Be sure: admit criteria never worked correctly. 100%
Kim
DG9VH

i2NDT, 2016-12-20:
any news on the "admit criteria" issue?

travisgoodspeed, 2016-12-20:
No news, but pull requests are always welcome.

i2NDT, 2016-12-20:
tnx Travis for the answer.
but it sounds strange to me that the admit criteria doesn't work! 
isn't it one of the mandatory features a dmr radio must fulfill to be accepted as DMR?

Randy-W0RDR, 2017-02-19:
My user group (about 41 people in Colorado) has noticed that the issue seems worse with the "tools" FW that it was on the original factory.  It's there on both.  We also noticed it was much worse on a MMDVM-Based repeater than on a genuine MotoTRBO (XPR 8300) - not sure if this helps with diagnosis...

eb4fbz, 2017-04-30:
Without a properly working admit criteria, sending periodic GPS frames will create interference on repeaters. This bug should be prioritary.

IZ4HUF, 2017-05-23:
Now there is a new firmware update from Tytera to solve the Admit Criteria, the new version is 013.34.
Please visit: http://www.tyt888.com/?mod=download
( find "MD-380 SOFTWARE 0321")

nivex, 2017-11-28:
On my radio the Admit Criteria is being respected w.r.t. a call in progress on another talkgroup. That is if I'm on Local and someone else is chatting away on TAC 310 (both on the same timeslot), I will get bonked when I try to key up.

Where this fails is if there is a call on my talkgroup with audio actively being processed, I can still key up right on top of it.

Of potential note: I just upgraded to CPS version 1.34 and there is a new channel setting `In Call Criteria` with options:
* Always
* Follow Admit Criteria

I have set this on an active channel here and sadly I was still able to key up over top of an ongoing call. My guess is it still needs some bits from the newer firmware mentioned by IZ4HUF.

For reference my MD380Tools version is 2017-06-11.

PD0DIB, 2017-11-29:
Think it 8s not custom Firmware related, that feature is handled by the
CPS.. I'm using CPS 1.37 and have in the Admin creteria Channel free and
the other is follow admin creteria. That's working ok
73
PD0DIB , Rob

====================
met vriendelijke groet | Kind Regards,
Rob van Rheenen

tristantech, 2018-07-24:
D013.34 is for MD380 models with the new vocoder (and seems to work for me), but does anybody have firmware for an old-vocoder MD380 that has the admit criteria fix?

i2NDT, 2018-07-25:
I am interested as well!!! 

 i2NDT's Web Site i2NDT's Grabber Compendium

JSB2000, 2019-04-24:
What's the fix status on this issue? My MD380 _was_ respecting the admit criteria setting on past updates of the firmware but just noticed that it was no longer doing so on recent versions.  

Currently running e0e885a CP ver 1.34 as pulled/installed with glv/glvflash on tytV4 virtualbox. No idea what "Dxxx.xx" version that's supposed to equate to.