travisgoodspeed/md380tools #123: VFO scan

dg9vh, 2016-05-18:
See https://groups.google.com/forum/#!topic/md380tools/efxqCRApBe4

travisgoodspeed, 2016-05-18:
We really need to have a full description and title for every issue; a teaser and link to the mailing list isn't enough.

Please edit this and the other "link only" issues to include (1) a thorough description of the request and (2) the exact condition when the issue can be closed as completed.

dg9vh, 2016-05-18:
Ok, i'll try to get it, if not till weekend, I'll close them.

dg9vh, 2016-05-18:
Add VFO menu or button and programmable control using the keyboard arrows 
Scan VFO - 136 - 174Mhz or UHF 403 - 470Mhz, Option to specify any frequency
from the keyboard in the VFO mode as in the meaning of direct frequency input.

travisgoodspeed, 2016-10-01:
Merged into #370. Reopen only if there is active progress in adding such a feature.

closed 2016-10-01 by travisgoodspeed