travisgoodspeed/md380tools #124: Adjustable keyboard sound volume, alert radios

dg9vh, 2016-05-18:
See https://groups.google.com/forum/#!topic/md380tools/efxqCRApBe4

dg9vh, 2016-05-18:
Volume of alert-tones, keyboard-tones etc. should be adjustable from minimum to maximum as number values from 0 to 100 as submenu in md380tools-menu called "sounds" or similar. I think it would be ok to make it controllable in general with one adjustment, if divideable into different controls it would be king-sized :smile: 

close if done or not realizeable

janipewter, 2016-07-21:
Yes please lock the tones volume to the master volume! It's near enough impossible to use this radio in quiet situations or with an earpiece without tones muted as you get deafened every time the call alert or channel free tones etc play!

eb4fbz, 2016-09-04:
Tones volume can't be modified with the master volume unless is done via hardware modification (https://github.com/travisgoodspeed/md380tools/issues/98). 
The MCU volume readouts are glitchy, you should not use it to scale the tones amplitude, it should be ignored.

travisgoodspeed, 2016-09-04:
Closing this issue, as we won't be able to do the change in software.  As @eb4fbz says, please see #98 for the hardware modification if you really need this feature.

closed 2016-09-04 by travisgoodspeed