travisgoodspeed/md380tools #126: Avoid the fact that if you turn the channel knob and you select an empty channel you hear an annoying sound, maybe locking in the last channel

dg9vh, 2016-05-18:
See https://groups.google.com/forum/#!topic/md380tools/zW7tkAydKRY

dg9vh, 2016-05-18:
If turning the channel knob to unprogrammed position a loud "duuuuuut" would be emitted. This should be muted and either the last channel selected should be selected with a short message displayed for lets say 5 seconds "channel not programmed, staying on last" or only the message "unprogrammed" should stay as in stock but without this noisy tone. 

Close if configurable via md380tools-menu to be switched on and of - some people need this tone to get on place (blinds)

travisgoodspeed, 2016-10-01:
Merged into #370. Reopen only if there is active progress in adding such a feature.

danny1990, 2017-06-06:
I found a fix it may not be the best but it works this also works for other sounds too.

in beep.c under "beep_OSMboxPend_hook()" set

case 15: // Invalid channel beep
    return 0;

KD4Z, 2017-06-07:
That did kill the unprogrammed channel tone.  However,  as you said, it also kills the "Roger Beep"  I made it optional in my VM to kick it around a bit.

travisgoodspeed, 2017-11-13:
Reopening.  We'll close this when #830 is merged.

PR #830 (2018-02-06): kill unprogrammed channel boooo tone
as title says. should fix #126

@dg9vh @KD4Z is this what you want?
closed 2018-02-06 by travisgoodspeed