travisgoodspeed/md380tools #127: Adding an option to adjust mic gain

dg9vh, 2016-05-18:
See https://groups.google.com/forum/#!topic/md380tools/zW7tkAydKRY

dg9vh, 2016-05-18:
Mic-gain should be adjustable from minimum to maximum as number values from 0 to 100, maybe assisted by @aeickho 's volume-meter in https://github.com/travisgoodspeed/md380tools/commit/989b8b08972385dedd72f77d8e73a5d640dcae80

Close if functional or dismissed.

der-stefan, 2016-05-25:
According to https://github.com/pchickey/md380-re/blob/master/datasheets/hrc5000.pdf (page 9) the C5000 has a register to set a MicVol. Or at least 4 bit to set 0dB, 6dB, 12dB or 20dB mic gain.

travisgoodspeed, 2016-09-23:
Merged into #370. Reopen only if there is active progress in adding such a feature.

closed 2016-09-23 by travisgoodspeed