travisgoodspeed/md380tools #129: Internationalize menu-items

dg9vh, 2016-05-18:
My idea is to replace the wordings of the menu-items in the firmware by getting them out of a txt-file with format:

address;old-menuitem;new-menuitem

So pre-required is to know the address and the wordings of the original menu-items to give the one who build his firmware the possibility to translate them into his language. The patching mechanism would catch those entries and write them into the firmware (if new-menuitem is not empty or to long), else puts out an error-messsage (for example: "String "..." to long" or so).

On this way, people could submit different <language>.txt files into the repository that others can use.

Close, if raw-file exists and patching could work with it.

der-stefan, 2016-05-18:
A structured XML file would be better than a unstructured txt file. The problem with a semicolon separated list is that a semicolon can never appear as a text. Perhaps we don't need it, but more professional is it in a structure.

terinjokes, 2016-06-08:
Chrome has an Internationalization JSON format for use with Chrome extensions that works pretty well. They define it here: https://developer.chrome.com/extensions/i18n-messages

travisgoodspeed, 2016-10-01:
Merged into #370. Reopen only if there is active progress in adding such a feature.

closed 2016-10-01 by travisgoodspeed