travisgoodspeed/md380tools #130: replace fw function pointer with function symbol

aeickho, 2016-05-21:
Hy travis,
with this functionality, we can save a "lot" of flash memory
see 6994e8c84955af0a6bb0f9df7cf6cbbe4c3d8fb5

the fw function is now called directly ...

 ```
 809e5dc:       461a            mov     r2, r3
 809e5de:       311c            adds    r1, #28
 809e5e0:       e88d 0060       stmia.w sp, {r5, r6}
 809e5e4:       f76e f8a4       bl      800c730 <_Min_Stack_Size+0x800c330>
 809e5e8:       4b07            ldr     r3, [pc, #28]   ; (809e608 <create_menu_entry_experimental_enable_screen+0x74>)
 809e5ea:       715c            strb    r4, [r3, #5]
 809e5ec:       b004            add     sp, #16
 
```
the address 800c730 is correct ... on armobjdump there is no +1

can I change globally?

travisgoodspeed, 2016-05-21:
Yes, please change it globally except where there's a good reason to use function pointers.

We should do the same with global variables, but that will need to be done more carefully to insure we don't introduce bugs.

aeickho, 2016-05-21:
function pointers <done>
global variables :neutral_face: I work on menu.c ....

travisgoodspeed, 2016-05-22:
Reassigned to @aeickho.

Close this issue when the function pointers and most of the global variables have been moved.  (No need to move all of them.)

PR #139 (2016-05-30): replace fw function pointer with function symbol
big modification of the code 
will be save ram and flash on the radio. 

Fixes #130
closed 2016-05-30 by aeickho