travisgoodspeed/md380tools #136: Add DMR IDs for Parrot etc. to users.csv

phl0, 2016-05-24:
The parrot function in the German Brandmeister net uses the DMR ID 262997 which is of course not contained in the user db on dmr.marc.net. Instead I would like to have it added by hand after downloading the DMR MARC db. 
Further work could include more echo and parrot TGs - probably beeing read from a file instead of inserting one after another. 
First part (yet to be tested) is here: https://github.com/phl0/md380tools/commit/9ed3b33f38f423bc1d9fe462b4eeba7ebe52fd10

travisgoodspeed, 2016-05-24:
DMRMARC's parrot and other talkgroups could also be added, but I'd rather store those the additions in a file than add bunch of echoes to the Makefile.

phl0, 2016-05-24:
Sure I think that is the better idea. Maybe people could give some hints on further DMR IDs used for parrots to be added. The commit in my branch is the first try. I will change that to include a file to walk through before I put up a PR for this.

dg9vh, 2016-05-24:
I think, it would be suitable to include also reflectornumbers, BrandMeister as also DMRplus because they are also announced as source when connecting to a reflector. Even 5000 as Status and 4000 as "not connected".

dg9vh, 2016-05-24:
Sources for reflector-Databases to be parsed and converted into user.db see
http://185.79.71.94/reflector.db for BrandMeister
and 
http://ham-dmr.de/reflector.db for DMRplus-Network.

phl0, 2016-05-24:
@dg9vh Tnx will include that.

dg9vh, 2016-05-24:
@phl0 take care of getting in conflict with names! It could be that DMRplus has some names for reflectors different than BrandMaster. Here I think (in fact to be more "international") we should overwrite those DMRplus-names with BrandMaster-Names.

phl0, 2016-05-24:
Hhhmm, I added 262997 as an example but it wouldn't show on the screen. Maybe the code is checking for DMR IDs to be at least 7 digits long?!

dg9vh, 2016-05-24:
@phl0 You have to put the line into users.csv in db-directory and not into the data - file... I tried it my way with following entry in users.csv and it works:
> 262997,PARROT,PAPAGEI,,,,,DMR

phl0, 2016-05-24:
@dg9vh That is exactly what I did. Will try a clean flash of the FW.

dg9vh, 2016-05-24:
I putt it on top of the file (at first line)

phl0, 2016-05-24:
@dg9vh That made the trick. I understand that the DMR IDs are sorted. So we will have to align these special IDs with what comes from DMR MARC.

dg9vh, 2016-05-25:
May the sort-command be with you, hi.

dg9vh, 2016-05-25:
> sort users.csv --field-separator=',' --key=1 > sortusers.csv

would solve our problem by sorting very quick ... Don't know if there is something similar in phython ...

phl0, 2016-05-25:
What is the source for this DB http://185.79.71.94/reflector.db for BrandMeister? I miss the TG26202 for example. Its a DMR->D-Star Bridge on XRF311 here. This is not contained...

dg9vh, 2016-05-25:
Source is http://bm-dmr.de/cp/pages/reflector.php - official BM-Germany list for DV4mini-inclusion with DV4MF2-Control-Panel and DL5KV's DV4Mini_compact-Panel

If something is missing, please mail to Meinhard, DL2MF to pull corrections

dg9vh, 2016-05-25:
@phl0 depending on http://wiki.bm262.de/doku.php?id=benutzerhandbuch#beschreibung_der_talk_groups I don't know if for TG26202 a such called "redirection reflector" for usage with DV4mini is established... what reflector-number is it?

phl0, 2016-05-25:
@dg9vh Tnx fer info. I mailed DL3OCK and DL2MF to ask about this issue. TG26202 is routed to D-Star Reflector XRF311 D. It does not have a DMR Reflector id from what I can see in the Wiki you mentioned.

dg9vh, 2016-05-25:
@phl0 based on this I think it could also not be possible to get TG26202 as "source of transmission" - even if there is a transmission from D-Star-side I think those would have CCS7-IDs, right? If not, I don't know what on DMR-side would be transported as source-ID.

If in such cases a number would be transported that is not mentioned in one of our source-paths given above in the URLs we have to catch the cat and maintain our own list for that... :-1: Don't know, if this would be fun

phl0, 2016-05-25:
I will test this issue to know which DMR ID is shown when I transmit to the D-Star side.

dg9vh, 2016-05-25:
@phl0 two tests would be nice:
one callsign with CCS7-ID and one with a callsign without.

phl0, 2016-05-25:
@dg9vh will do

dg9vh, 2016-05-28:
@phl0 How far are your experiments with the issue? Still any problems?

phl0, 2016-05-28:
@dg9vh Not had time yet, due to http://ballon.n18.de :-)

dg9vh, 2016-05-28:
@phl0 Nice project :+1:

dg9vh, 2016-06-03:
@phl0 If starting merging the lists into the database, please consider about following database containing talk-groups for dv4mini-usage.
http://195.225.116.10/ts1.db

if you switch to a TG in TS1 via DV4mini the ControlPanel announces with source id out of the list above the TG switched to...

dg9vh, 2016-06-03:
http://195.225.116.10/ts1.db is also ham-dmr.de/ts1db

phl0, 2016-06-22:
Got my MD380 back. Will go on with the work for this the other day.

travisgoodspeed, 2016-06-22:
With the pull request for this issue, please be sure that:
- [ ] The code runs on BSD and OS X.  Specifically, use perl instead of grep or sed for transformations, as BSD's text tools can't handle Unicode.
- [ ] The extra databases don't damage lookups from the old database.  Maybe check for duplicates or collisions?

Looking forward to closing this issue soon.

phl0, 2016-06-29:
I have conducted a few tests with D-Star Reflector XRF311 D linked to TG 26202 on BrandMaster network. If I transmit D-Star on XRF311 D with my own Callsign DF2ET, CCS7 ID 2624112 it appears with the CCS7/DMR ID on the BM network and hence on the MD-380 screen. So this is not an issue here.

If I use another call (DN1YWF for example, registered to me for educational puporses) the source ID in BM network is 26202 so the talk group as ID.

[image]

dg9vh, 2016-06-29:
Did you put the ID in logical order of the DMRid-database-file?

phl0, 2016-06-29:
At the the moment its the "stock" DMR ID database. So no additions yet. It was just to answer your questions from May 25th :-)

dg9vh, 2016-06-29:
"Was kümmert mich mein Geschwätz von gestern!" 

Well, ok, now I got the reference... the question was, what happens if one comes with a non-registerted callsign via D-Star. 

So now it's clear for us to get all talkgroups into the database... there would be space enough even for TGs never be needed... would be easier to get via URL above...

phl0, 2016-06-29:
Ack ack, I just tried with 5000 for status info and 4000 for disconnect. These two come back with 4000 resp. 5000 as ID. So I suggest that we add these two as a first step.

phl0, 2016-06-29:
Linking to other reflectors results in a Transmission showing the reflector ID as source ID. I just tried 4400 for example.
[image]
This was tested on my DVmega with MMDVMHost connected to the German BrandMaster network. As it was announced today that the reflectors 4000 - 4999 from DMR+ and BM are now interlinked, I guess we should take the BM list as source. Any suggestions?

phl0, 2016-07-21:
Ok, as of today the interlink of specific reflectors between DMR+ and BM has been disconnected again. DMR-MARC has been interlinked with some special TGs. For DMR+ there is currently only three specific reflectors (in Germany) that are linked to BM. So I will go for the list from BM for now.

phl0, 2016-07-25:
I created a PR for this: https://github.com/travisgoodspeed/md380tools/pull/199
What do you think?

travisgoodspeed, 2016-07-26:
Merged #199, 8d3d9b3.  Thanks, Florian!

Anyone know where to get a database of the DMRMARC talkgroups?  I'd like to include them and finally close this issue.

phl0, 2016-07-26:
I will have some research on this. Pls QRX

phl0, 2016-07-26:
Would have another issue. At least in Germany we have a parrot fucntion on TG262997. I would like to add this as well. Maybe put 5000 and 262997 into a file "special_TG.csv" or the like?
And does anybody know if there is something similar to the German parrot in other countries / TGs?

dg9vh, 2016-07-26:
@phl0 I post a question into the brandmeister-yahoo-group and hope we get info about this.

phl0, 2016-07-26:
Just made a quick test on a DMR MARC repeater. As TGs like 2624 (for regional purposes in Germany for example) only appear as destination they are currently not of any use because the info screen on the MD-380 only shows information about source IDs/TGs. Have I missed something?

cyanide-burnout, 2016-07-26:
Parrot on BM is not TG, it is private call.

phl0, 2016-07-26:
Okay sorry. But it is listed with source ID in MD-380 when it returns the audio. So it would make sense if we have it in the user DB.

@cyanide-burnout: Are there more parrots than 262997 out there?

cyanide-burnout, 2016-07-27:
Yes, each country has its own ID for parrot. It depends on national administration choice.

dg9vh, 2016-07-27:
Forwarded from brandmeister-yahoo-group:

> You can try 425997 on brandmeister-il.ddns.net
> The demo app is on 425991

Just doing further collection.

@phl0 using a special_tg.csv would be a very good idea.  But we have to keep in mind, that the final list has to be sorted in accending way I think... so when merging, we keep to be sure that they are sorted?!

phl0, 2016-07-27:
Is there an overview (online) of which IDs / TGs are currently assigned for such purposes?

cyanide-burnout, 2016-07-27:
We have no such service right now, but this is good idea to make method to get it on APIs of each servers.

phl0, 2016-07-27:
> We have no such service right now, but this is good idea to make method
> to get it on APIs of each servers.

Okay. Where can I place a feature request please?

bilbolodz, 2016-07-27:
I think that good idea is to add "local file" (edited by user) that will be merge during "make flashdb" with remote (server) content. That should solved problem with "local national services" and user tweaks.

cyanide-burnout, 2016-07-27:
> Okay. Where can I place a feature request please?

It is already placed to internal tracking system :)

phl0, 2016-07-27:
> It is already placed to internal tracking system :)

Tnx Artem. Please keep me posted about this issue. Or is the status of
the feature request publicly available?

phl0, 2016-07-27:
I made another PR (see https://github.com/travisgoodspeed/md380tools/pull/201) because there was a bug with a missing newline between the data pulled from BM and the data pulled from DMR MARC db. 
Afterwards I made a separate file containing all special IDs / TGs that can currently not be pulled from any location on the net. Sorting is not necessary up to now as long as the file special_IDs.csv is kept sorted. But I will keep the isse about sorting in mind for further work on this topic.

cyanide-burnout, 2016-07-27:
Test example - http://aesyle.dstar.su/api/service.php
Is it OK?

phl0, 2016-07-27:
@cyanide-burnout data look ok. But for md380tools we would need the data comma separated. Would it be possible to provide the data also in csv format?
See here for example: https://raw.githubusercontent.com/phl0/md380tools/dev/db/special_IDs.csv

cyanide-burnout, 2016-07-27:
Uhh.. And you want to get it aggregated.
So please wait for a while. We need to create aggregation service on the main portal. Our aggregated API's developer @zarya is on vacation :)

phl0, 2016-07-27:
@cyanide-burnout np at all. In the mean time we can live with a separate file. I will move it to automatically pull the data from bm as soon as you finished. Take your time and thank you :)

cyanide-burnout, 2016-07-27:
I know the simplest way :) You also want to have text description. That is possible in one case only: we should special_IDs.csv to each master server. The file will be edited my local administration of each server.
Server list is available at http://registry.dstar.su/api/node.php and http://registry.dstar.su/dmr/dmrmaster.php 
The pool URL will be http://<server address>/md380tools/special_IDs.csv
Are you agree?

phl0, 2016-07-27:
That list shows the masters per country as far as I can see. So you want to set the column containing the country according to the master number of each master server?

cyanide-burnout, 2016-07-27:
.. Or give the user chose country and server and then inport right list from particular server.

phl0, 2016-07-27:
Currently we have enough space in the MD-380 radios. So I would prefer a centralized approach without any user interaction needed to select the server.

[3 more replies]

closed 2016-08-16 by aeickho